adafruit/circuitpython #163: neopixel_write broken?

mindforger, 2017-07-12:
i just updated to 1.0.0 RC4 and wanted to take some measureing for power consumption ... but i am unable to turn off the status LED

>>import neopixel_write
data = bytearray([0,0,0])
neopixel_write.neopixel_write(baord.D8, data)

it is just changing from green to white and then blinks blueish until it repeats

tannewt, 2017-07-12:
That's indicating an error. You have a typo with baord instead of board.

mindforger, 2017-07-12:
i did not copy that, i had a typo while writing the issue, this is NOT the issue
if you read my edited post i need an working example, because the documentation is very inconclusive on how to use the command and it's signature

mindforger, 2017-07-12:
yeah i just noticed you answered via email, so you were not able to read it, the used code is absolutely fine, i edited it line for line, only the neopixel_write always fails!

tannewt, 2017-07-12:
I'll take a look at providing better docs and example later today. What docs are you currently looking at? I expect most people to use the [NeoPixel library](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel) instead of `neopixel_write` directly.

mindforger, 2017-07-12:
I tried that one and it failed right at the last line call to neopixel_write

tannewt, 2017-07-12:
You tried the library and it failed? Please post your code.

I also put an example here: https://circuitpython.readthedocs.io/en/stable/shared-bindings/neopixel_write/__init__.html

jerryneedell, 2017-07-12:
Scott- I tried you example on an M0 Express - 2 issues 
I think the OUTPUT has t be Upper Case
Then the last line failed
This was with 1.0.0.rc4-1  adn the lastest librry bundle 1.0.0-rc2 release
Did I miss an update?

>>> import board
>>> import neopixel_write
>>> import digitalio
>>> pin = digitalio.DigitalInOut(board.NEOPIXEL)
>>> pin.direction = digitalio.Direction.OUTPUT
>>> pixel_off = bytearray([0, 0, 0])
>>> neopixel_write(pin, pixel_off)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'module' object is not callable

- Jerry

tannewt, 2017-07-12:
Yeah, I thought I had fixed that. It should be `neopixel_write.neopixel_write(pin, pixel_off)`

jerryneedell, 2017-07-12:
Cool! That works!

jerryneedell, 2017-07-12:
And it does have to be:
pin.direction = digitalio.Direction.OUTPUT
correct?

tannewt, 2017-07-12:
yeah, I don't know why thats not fixed either

tannewt, 2017-07-12:
Not saving in my editor would do it. *groan*

closed 2017-07-12 by tannewt

mindforger, 2017-07-13:
oh come on, i need to set the direction of the pin before using it and that is why it stopps running XD
i thought the syntax was wrong in some way

now everything works just fine, thank you

tannewt, 2017-07-13:
@mindforger it defaults to input. Thats one reason why using the neopixel library is easier. :-)

mindforger, 2017-07-13:
@tannewt i must have used a bit different version of the neopixel lib but the one you linked won't even import without green -> yellow(blueish when i use neopixel.py instead of neopixel.txt) -> 5x blueish
>from neopixel import *

[neopixel.txt](https://github.com/adafruit/circuitpython/files/1144137/neopixel.txt)

tannewt, 2017-07-13:
So are you all set?

mindforger, 2017-07-13:
well i am in a learning process to find all the hidden details i have to look out for when not using an dedicated IDE ... i am spoiled ... maybe i should return back to writing stuff in a notepad and compile it by hand for a bit basic practise again

commit a91592df (2017-07-12): shared-bindings: Add neopixel_write example. Fixes #163
commit 5e231bd8 (2017-07-12): shared-bindings: Add neopixel_write example. Fixes #163
commit b69507fc (2017-07-12): shared-bindings: Fix neopixel_write example. I forgot to save in the editor. Fixes #163
commit cc0cf717 (2017-07-12): shared-bindings: Fix neopixel_write example. I forgot to save in the editor. Fixes #163