adafruit/circuitpython #164: how does one open or create a file for writing?

mindforger, 2017-07-13:
i tried 

>f = open("test.txt","r")
f.close()

and it runs

but using 

>f = open("test.txt","w")
f.close()

i only get an orange LED and cyan blinking as a result

is it not implemented yet? or did i miss some kind of detail again?

I want to store a few data that i can later extract easily through USB

tannewt, 2017-07-13:
Sorry I'm going to be terse but I'm on my phone and then sleeping. I just
answered this on the forum:
https://forums.adafruit.com/viewtopic.php?f=60&t=120169

It also sounds like you aren't connected to the repl. Doing so should
provide more detailed error messages for you. More info here:
https://learn.adafruit.com/micropython-basics-how-to-load-micropython-on-a-board/serial-terminal

Thanks and good night!

mindforger, 2017-07-13:
ahaaa okay, i will look into that then and close this for now

mindforger, 2017-07-13:
mh unfortunately i can not build 2.0.0alpha ... i seem to miss a lot of dependencies ... moved over the git to my local win8 pc, setup the compiler and ran "make BOARD=feather_m0_express" but it fails on "genhdr" looks like i require python to be present or something but i don't understand it

dhalbert, 2017-07-13:
@mindforger What guide did you use to set up the dev environment on Windows? There are several ways to get started:

1. Use Cygwin or MinGW. This will be difficult to get right.
2. Use vagrant to get a [preconfigured Linux VM for development](https://learn.adafruit.com/micropython-for-samd21/build-firmware?view=all#build-firmware ). [Here's a general overview](https://learn.adafruit.com/a-quick-linux-vm-on-windows-with-vagrant/overview ) of vagrant. There are some [potential stumbling blocks on Windows](https://forums.adafruit.com/viewtopic.php?f=60&t=109632).
3. Use Bash for Windows on Windows 10. See [this thread](https://forums.adafruit.com/viewtopic.php?f=60&t=112975).  Works but the compile time is a little slow (2 mins vs 30 secs).
4. Use VMWare Player or VirtualBox and install regular desktop Ubuntu. Then install a few extra packages. Use the list described [here](https://forums.adafruit.com/viewtopic.php?f=60&t=112975&p=564651#p564651).
5. Build a standalone Ubuntu machine or make your Windows PC be dual boot. I have found this to be the easiest in the long run, if you have the hardware to do it. Dual-booting on a single drive is a bit tricky because major Windows upgrades can wipe out the grub bootloader, which replaces the Windows bootloader. I use a second hard drive instead and choose the boot drive via the BIOS boot menu.

The missing `freetouch_ptc.c` is due to a step you need to do to load external "submodules" via git. These are external packages used in the build but not included. You only need to do this once.
```
git submodule init
git submodule update
```

mindforger, 2017-07-13:
i tried using AVR gcc that i use for my normal ATSAM/feather stuff
VM is a nogo on this machine and cygwin clashes heavily with the AVR compiler
did not test yet with MinGW
Win 10 is not available only Win8 for the same reasons a VM is a no-go
dual-boot the same ... i could setup a RPi as building machine :D 
will give it a shot this weekend ... until then i will have to think on how i can log my data with other hardware

dhalbert, 2017-07-13:
RPi is possible. It will be slow but will work. We have someone doing that. I have done dual-boot with an external USB3 drive and it was perfectly fine. But if it's a work restriction I can understand that.

mindforger, 2017-07-13:
USB boot ... nice idea! Will try that out ... any recommendation on distro ?

dhalbert, 2017-07-13:
I use Ubuntu, and use the xenial 16.04 long-term release. It's great. Follow the instructions [here](https://forums.adafruit.com/viewtopic.php?f=60&t=112975#p564651) re the packages you need. Bash on Windows _is_ Ubuntu, so the instructions are identical.

To use `screen /dev/ttyACM0` to talk to the REPL, you will need to change some permissions. You can follow the instructions [here](https://learn.adafruit.com/adafruit-arduino-ide-setup/linux-setup) about `udev` rules. Or you can run `screen` as `sudo` in the meantime.

Gotta go for a few hours but will check back later.

tannewt, 2017-07-13:
I recommend ubuntu. Debian arm gcc is very old.

tannewt, 2017-07-13:
If you don't intend on changing the CircuitPython core, then you can just use prebuilt binaries. Travis CI builds every commit and places them here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_m0_express/ Just match up the commit hash with the commit hash you see in GitHub and you should be set.

I'm going to close this because there is no action to take for us. However, feel free to follow up here, on Discord (http://adafru.it/discord) or on the forums.

closed 2017-07-13 by tannewt

mindforger, 2017-07-14:
Yeah sure, didnt know you have builds public, nice