adafruit/circuitpython #165: unsure if bug or just windows or wrong driver

mindforger, 2017-07-13:
when i install the arduino drivers for COM ports to talk to my feather m0 express, the mass storage vanishes again under win8 (like before i updated the bootloader, it is just not existent),  it seems i can not have both or do i need to modify or write soemthign special to the boot.py or code.py to accomplish both uart for remote debug printout and mass storage for editing?

tannewt, 2017-07-13:
Its a Windows thing but I'm not sure what the right combination of drivers is. @dhalbert would know. Also, for technical support like this, please start at the forums: https://forums.adafruit.com/viewforum.php?f=60 (I get emailed about new posts there too.)

Once we know its a CircuitPython bug we can create an issue.

closed 2017-07-13 by tannewt

dhalbert, 2017-07-14:
Yes, open a forum post. But try uninstalling all the "Adafruit" drivers and then reinstalling. I've seen that fix missing COM ports, including on my own machine.