adafruit/circuitpython #177: built in samd temperature sensor

ladyada, 2017-07-30:
there's a 'calibrated' temperature sensor that seems to work pretty well. see the START example and http://www.atmel.com/images/Atmel-42645-ADC-Configurations-with-Examples_ApplicationNote_AT11481.pdf

non-ASF arduino code attached. mostly its getting the NVRAM data out and then tweaking the ADC to read the TEMP input w.r.t the 1.0 vref

would be nice to compactify and stuff into gemma/trinket core because it would give people a sensor to play with :)

[inttempsensor.txt](https://github.com/adafruit/circuitpython/files/1185911/inttempsensor.txt)

tannewt, 2017-08-08:
I think we should add a `die` attribute to the `microcontroller` class which is an object with a `temperature` property. That way you can read it with `microcontroller.die.temperature`.

ladyada, 2017-08-08:
okidoke!

dhalbert, 2017-08-08:
Is there anything else that would go under die? If not, just die_temperature and avoid the new object?

ladyada, 2017-08-08:
sometimes you can measure on-die voltage (e.g. vcc) but i haven't worked out all the tricks on samd for that yet :)

ladyada, 2017-08-08:
also maybe die.speed (set/get running freq)?

dhalbert, 2017-08-08:
Current definitions in `microcontroller`:
`.delay_us(t)`
`.disable_interrupts()`
`.enable_interrupts()`
`.Pin`  [class]
`.pin`   [module, defining pins `.PAxx` and `.PBxx`]

So `die` looks good, kind of parallel to `pin`. Sorry I doubted!

tannewt, 2017-08-08:
No problem! I'm happy to have API discussed.

One tricky point is that its not possible to put properties at the top level in a module. They must be on an object. So, `microcontroller.die_temperature` would have to be a function call or wrapped in an object. Having it wrapped in an object can be nice because it can be passed around then like any other temperature sensor (they all should have a `temperature` property).

tannewt, 2017-08-21:
@brentru Do you have time to do this? If not, please unassign yourself.

dhalbert, 2017-08-30:
Implemented as `microcontroller.core.temperaure`. A few notes:

It works! There is significant thermal mass, but reflects the ambient temperature pretty well. Just putting your thumb on the chip doesn't do much immediately., but breathing on it will raise the temperature after a few seconds. I also put it in the freezer and it tracked nicely. Tested on CPX, Gemma M0, and Trinket M0.

Temperature value can vary by several tenths of a degree Celsius between consecutive readings. I am using the ADC parameters given in the application note. If I set up the ADC to take many more readings and do averaging, the value becomes much more stable.

As noted in the application note, the first reading is pretty different than subsequent ones. I read twice and discard the first. Interestingly, I don't see the same effect with `analogio.AnalogIn`: the first reading there is close to or the same as the second. The temp settings use the 1V reference; `AnalogIn` uses Vcc.

Despite the factory calibration values, two samples of the same board can vary by a couple of degrees or more.

For debugging reasons, I implemented `core.frequency` as well. I left that in, and it's available in both the 
SAMD and ESP8266 ports. ESP8266 doesn't have a public temperature API, so it returns `None`.

ladyada, 2017-08-30:
yep this isnt precision, its just sort of a 'hey you get a free sensor for playing with' type thing :)

PR #211 (2017-08-31): Add microcontroller.core.temperature, for use as a simple sensor on minimal boards.
[replaces #210 due to force push problems]

Add microcontroller.core.temperature, for use as a simple sensor on minimal boards.

Also added microcontroller.core.frequency to return CPU frequency, in int Hertz.

Added -finline-limit=_n_ to atmel-samd/Makefile to squeeze non-Express enough to fit.

This branch crossed the v1.9.2 merge, so I gave up trying to rebase.

Fixes #177.
closed 2017-08-31 by dhalbert