adafruit/circuitpython #184: Add a way to do a reset from code

deshipu, 2017-08-08:
Sometimes the code might need to do a reset, for example, as a way of recovering from an unexpected exception in a long-running program, or as a way of clearing the state of the device to load a new subprogram.

mmabey, 2017-09-18:
@deshipu I don't know if these are the kind of solutions you're looking for, but MicroPython has a [`reset()`](http://docs.micropython.org/en/latest/esp8266/library/machine.html#machine.reset) function as well as something called a [watchdog timer](http://docs.micropython.org/en/latest/esp8266/library/machine.WDT.html). Perhaps @tannewt can comment on the likelihood that one or both of these could be ported over to CP.

Andon-A, 2017-11-27:
This would be super-useful in a few of the examples that I am planning for #345

tannewt, 2017-12-05:
You mean to reload the program right? A reset would do a fully reinitialization of the microcontroller and stop USB when available.

deshipu, 2017-12-05:
I mean the same thing that ctrl+d does in REPL.

tannewt, 2017-12-05:
OK, I call that reload.

The tricky part here is that we want to exit the existing code and immediately re-run it. Normally, we'd go into our wait state with the status led fading in and out.

caternuson, 2018-03-26:
It'd also be useful if a file to run on reset could be specified. This would let people use this in main.py to specify a different file to run and eliminate the need to rename. Or it could be used in a more complex way like a menu system to allow selecting one of the other .py files to be run programmatically.

tannewt, 2018-04-02:
@deshipu thats a great idea! Mind filing a separate issue for it?

rhooper, 2018-05-14:
I'm going to have a look at this.

tannewt, 2018-05-16:
Fixed by @rhooper in #838. Thank you!

closed 2018-05-16 by tannewt