adafruit/circuitpython #185: Timing differences when serial is attached to different USB ports

dhalbert, 2017-08-10:
This code https://gist.github.com/sethcohn/0d2493dc2b854bd5584fe09fc0cd8509 shows timing differences when the USB port is attached to different USB ports (major differences USB2 vs. USB3, minor differences on various USB2 ports).

Dell 7010, has USB2 and USB3 ports:
USB2: typically 0.25 secs per loop iteration
USB3: typically 0.125-0.127 secs per loop iteration
(using Tera Term on Windows, `screen` on Linux)

USB2, old Mac Airs (2010, 2011):
May start out with 0.1x, but rises to 0.24 - 0.30. High variance.

USB3, new Mac (Scott's machine):
Similar to Dell 7010 USB3

sethcohn, 2017-08-10:
USB 2, screen, Linux, similar to Mac Airs.

Happy to help :)

tannewt, 2018-01-16:
Whats the goal with this? We make no timing promises when running python code. It should be expected to be variable.

tannewt, 2018-07-03:
@dhalbert want this open?

dhalbert, 2018-07-03:
I tried this test with 3.0 and no longer saw a significant variance. So I'll close.

closed 2018-07-03 by dhalbert