adafruit/circuitpython #186: [SUGGESTION] add the possibility to detect if the board is connected as mass storage or just usb powered

mindforger, 2017-08-10:
like the title says, it would help to have a way to detect if the device is connected only on an usb power source or actually is accessed as a mass storage device

this way i could write myself a boot.py that could wait around 5seconds before actually calling my main.py
if the device is connected as mass storage, i would prevent causing damage to the internal writing system but if the device is not connected as mass storage i would disable the mass storage interace after 5 seconds and allow self writing

tannewt, 2017-08-10:
I'm not sure how we'd be able to do this exactly because USB isn't started until after boot.py finishes. One tip for making a logger is using a physical switch or button to read at boot and determine which mode to boot into.

mindforger, 2017-08-10:
oh shit i did not know that, sorry :(

yeah i did that with the jumper but i thought it could be made easier that way :)

closed 2017-08-10 by mindforger

tannewt, 2017-08-10:
No worries! I think it's a super common thing that will come up and don
from discord came up with the switch idea which is a good one!