adafruit/circuitpython #187: digitialio -document how to unset pullup/down

ladyada, 2017-08-11:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/digitalio/Pull.html

'None' is valid too! :)

closed 2017-08-22 by dhalbert

kevinjwalters, 2020-12-14:
[Adafruit Learn: CircuitPython Basics: Digital Inputs & Outputs: Digital Inputs](https://learn.adafruit.com/circuitpython-digital-inputs-and-outputs/digital-inputs) mentions `digitalio.Pull.NONE`. I put some feedback in as that does not exist and I'm not sure it ever has, doesn't appear in CP 2 docs.

dhalbert, 2020-12-14:
> [Adafruit Learn: CircuitPython Basics: Digital Inputs & Outputs: Digital Inputs](https://learn.adafruit.com/circuitpython-digital-inputs-and-outputs/digital-inputs) mentions `digitalio.Pull.NONE`. I put some feedback in as that _does not exist_ and I'm not sure it ever has, it doesn't appear in CP 2 docs.

Fixed in the guide. Thanks for spotting this.

PR #188 (2017-08-22): shared-bindings: Update DigitalIO.pull docs and validate that third state.
That way other values besides None throw errors instead of acting like None.

Fixes #187
commit f570b53d (2017-08-22): shared-bindings: Update DigitalIO.pull docs and validate that third (#188)

state is None and not anything.

Fixes #187
commit d7fae0b7 (2017-08-23): shared-bindings: Update DigitalIO.pull docs and validate that third (#188)

state is None and not anything.

Fixes #187