adafruit/circuitpython #197: missing uasyncio libraries

lila, 2017-08-20:
my understanding is that the current version of [circuit python (v1.0)](https://github.com/adafruit/circuitpython/releases/tag/1.0.0) is based on micropython 1.8.7.  micropython (i think) does support uasyncio and has a external libraries project called [micropython-lib](https://github.com/micropython/micropython-lib) that holds a number of librariies including the uasyncio library (there are two there, one is uasyncio which is empty and uasyncio.core which is what i want).  

i guess my question is:  how can i build/download a compatible version of uasyncio.core for use with circuitpython v1.0?

i can package the contents of uasyncio.core and copy it to circuitpython's lib folder, but when i try to define an async function, it generates a syntax error.  for [example](url):

`async def pwm_cycle(led, duty, cycles):`

The error is:

> File "code.py", line 6
> SyntaxError: invalid syntax

 any pointers are appreciated...

thanks,

- k

lila, 2017-08-20:
just more details:

it appears that the `async` keyword is new in python starting at version 3.5.  circuiltpython v1.0 is using python version 3.4.0.  

>import sys
>print(sys.version_info)
>(3, 4, 0)

but 3.4 does have coroutines, using `@asyncio.coroutine` instead of `async`.  however, 

>import asyncio
>Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>ImportError: no module named 'asyncio'

- k

tannewt, 2017-08-21:
What board are you trying this on? I'd expect it to work on ESP8266.

lila, 2017-08-21:
i'm using the [Adafruit Circuit Playground Express](https://www.adafruit.com/product/3333).

tannewt, 2017-08-21:
Ok, I've not taken any time to look into adding it. What libraries are you trying to use on top of it?

willingc, 2017-08-21:
@lila The async stuff in MicroPython and its additional libraries is a subset of CPython and still looks a bit work in progress. What's your use case for the module?

lila, 2017-08-21:
i'd like to use libraries such as :  https://github.com/peterhinch/micropython-async  in order to do event driven programming of the board.  For example, when buttons are pressed or accelleratometers triggered do run a coroutine, rather than have to explicitly poll for each of the inputs.

in https://github.com/peterhinch/micropython-async/blob/master/aswitch.py  there is a pushbutton class that handles the registration and execution of the python functions.  

i find that's a much easier way to program...  

if there are other ways to do this, please let me know...  thanks,

- k

tannewt, 2017-08-21:
I think async programming is much harder to teach than sync programming so this is very low on my list.

If you or anyone else are interested in adding support, please reopen for tracking. Otherwise, I'm closing this. While I think we'll get there eventually, there are just too many other things to do including additional MCU support, standardizing APIs and networking to consider this in the medium term.

closed 2017-08-21 by tannewt

sethcohn, 2017-08-21:
The micropython uasyncio code is very actively being worked on, as in a commit that is only a day old (as of now), by pfalcon.  Until it's reasonably stable, I can't see that it's gonna get much attention for the limited number of ports (samd, esp8266) that adafruit folks are trying to get into a manageable shape. (and keep in mind, CP doesn't track MP exactly, there are some differences such as using non-micro libraries (random vs urandom, for one example...))

ronysh, 2018-03-04:
I totally agree that event driven input can bring working with these boards to the next level.

tannewt, 2018-03-05:
@ronysh please follow #619 for more info.

rosstie, 2022-06-03:
So just to clarify asyncio is not currently supported on Circuit Playground Express? I have been trying to work this out as I would like to utilize cooperative multitasking.

dhalbert, 2022-06-03:
@rosstie That's right, there just isn't room for it in the firmware space we have (256kB, which includes several frozen libraries).

ladyada, 2022-06-03:
you could try the circuit playground bluefruit instead! it has a ton of flash/ram :)