adafruit/circuitpython #204: Move `math` into shared-bindings/shared-module.

tannewt, 2017-08-24:
This will improve its documentation and structure.

tannewt, 2017-09-19:
Two examples of commits that do a similar thing are: https://github.com/adafruit/circuitpython/commit/a0058e67124b93eb1c8c9c610f823396a7ff7e68
https://github.com/adafruit/circuitpython/commit/778e975936b9e892c4f1b85ce73462a0ce6d36d2

Most of the work is restructuring the code and verifying the API is a strict subset of [CPython's struct](https://docs.python.org/3/library/math.html). Do NOT copy documentation from CPython.

If you have any questions, please visit the #circuitpython channel on [Discord](http://adafru.it/discord).

mrmcwethy, 2017-10-20:
The following functions are supported by Micropython, but currently are NOT supported in CircuitPython.  Is this a surprise?  To save memory I suspect.  These function are not commonly used.
Functions
---------

.. function:: acosh(x)
   Return the inverse hyperbolic cosine of ``x``.
.. function:: asinh(x)
   Return the inverse hyperbolic sine of ``x``.
.. function:: atanh(x)
   Return the inverse hyperbolic tangent of ``x``.
.. function:: cosh(x)
   Return the hyperbolic cosine of ``x``.
.. function:: erf(x)
   Return the error function of ``x``.
.. function:: erfc(x)
   Return the complementary error function of ``x``.
.. function:: expm1(x)
   Return ``exp(x) - 1``.
.. function:: gamma(x)
   Return the gamma function of ``x``.
.. function:: ldexp(x, exp)
   Return ``x * (2**exp)``.
.. function:: lgamma(x)
   Return the natural logarithm of the gamma function of ``x``.
.. function:: log10(x)
   Return the base-10 logarithm of ``x``.
.. function:: log2(x)
   Return the base-2 logarithm of ``x``.
.. function:: sinh(x)
   Return the hyperbolic sine of ``x``.
.. function:: tanh(x)
   Return the hyperbolic tangent of ``x``.

tannewt, 2017-10-20:
Are they turned off by a #define? If so, its probably just an oversight. @dhalbert would have an opinion about whether to include them because they may increase code size.

I'm ok leaving them out until we hear from folks that they need something.

mrmcwethy, 2017-10-20:
the functions that are not in CircuitPython are surrounded by #if MICROPY_PY_MATH_SPECIAL_FUNCTIONS/#endif.

dhalbert, 2017-10-20:
[comments crossed]

These are turned on by `#define MICROPY_PY_MATH_SPECIAL_FUNCTIONS (1)`. If I turn them on, they use up an extra 15k flash. So they're not so desirable, even for the M0 Express boards.

We could turn them on in the M4 build, maybe, though we'll need to see what our flash budget is.

Note that complex number support is turned off on the non-Express M0 builds right now. That also is large enough to make a difference.

mrmcwethy, 2017-10-20:
The embedded documentation should reflect which math functions/features are available on which boards.  For example we could put a disclaimer in like "The following functions are supported on all boards: <math function list>.  The following functions may not be available depending on your board: <match function list>.  The determine what functions are available enter the REPL type import math followed by dir(math) "  

The documentation would, of course, be simpler if all boards supported the exact same list of functions.

tannewt, 2017-10-23:
Fixed by @mrmcwethy in https://github.com/adafruit/circuitpython/commit/bb008cdbc963ed3c3bdb949c7f7b6825d000ab40. Thanks!

closed 2017-10-23 by tannewt