adafruit/circuitpython #205: Refine `ustruct` into `struct` in shared-bindings and shared-module.

tannewt, 2017-08-24:
Again, good for docs and structure.

tannewt, 2017-09-19:
Two examples of commits that do a similar thing are: https://github.com/adafruit/circuitpython/commit/a0058e67124b93eb1c8c9c610f823396a7ff7e68
https://github.com/adafruit/circuitpython/commit/778e975936b9e892c4f1b85ce73462a0ce6d36d2

Most of the work is restructuring the code and verifying the API is a strict subset of [CPython's struct](https://docs.python.org/3/library/struct.html).

If you have any questions, please visit the #circuitpython channel on [Discord](http://adafru.it/discord).

mrmcwethy, 2017-09-28:
I was just comparing Circuitpython ustruct with Python 3.5.2 struct and ran the follow tests.  In Python 3.5.2, the @lLlL pack format and the lLlL format produce 8-byte integers while <iLIL and >iLiL format produces 4-byte integers.  Circuitpython produces 4-byte integer in all cases.  I think that the native size of long on my Ubuntu machine must be 8-byte integers therefore this  difference should be allowed.

Mike

>>> struct.pack("@lLlL",1,2,3,4)
b'\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00\x00\x00\x00\x00'
>>> struct.pack("lLlL",1,2,3,4)
b'\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00\x00\x00\x00\x00'
>>> struct.pack("<lLlL",1,2,3,4)
b'\x01\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x04\x00\x00\x00'
>>> struct.pack(">lLlL",1,2,3,4)
b'\x00\x00\x00\x01\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x04'

tannewt, 2017-09-28:
How do the docs define what the result should be? Is it platform dependent?

mrmcwethy, 2017-09-28:
The Python 3.6 docu (section [7.1.2.2.](https://docs.python.org/3/library/struct.html?highlight=struc#module-struct)) says "When a format string that starts with @ character or no special char, the native size is the used.  When using native size, the size of the packed value is **platform-dependent.**".  If the format string starts with '<', '>', '=','!' the number of bytes is specified in a table (i.e. not platform specific).

mrmcwethy, 2017-09-29:
I have another difference in behavior Python 3.5.2 struct and CircuitPython 2.0 ustruct.  See below the same code is on both systems.  I was being a little tricky and and set the offset to 1 instead of 0.  The Python 3.5.2 case placed the short int right at the offset=1, while CircuitPython placed it at offset=2!  I am not sure why, but I am guessing the CircuitPython is forcing packing alignment from the beginning of the buffer,while Python 3.5.2 does the alignment starting at offset=1

**Python 3.5.2**
>>> b=bytearray(b'\xff\xff\xff\xff')
>>> struct.pack_into('h',b,1,1)
>>> b
bytearray(b'\xff\x01\x00\xff')

**CircuitPython 2.x**
>>> b=bytearray(b'\xff\xff\xff\xff')
>>> struct.pack_into('h',b,1,1)
>>> b
bytearray(b'\xff\xff\x01\x00')

dhalbert, 2017-09-29:
Is this relevant? https://github.com/micropython/micropython/issues/3314 (_"ustruct unpack_from with nonzero offset doesn't match CPython (esp8266)"_). Our code is the same as MicroPython.

mrmcwethy, 2017-09-29:
Good question.  Is this why Micropython called the class ustruct or was it something else.  The good news is that Micropython ustruct.packinto() and CPython struct.pack_into() do have the same signature. And they function very much the same.   Also since the native formats are  platform dependent, maybe this edge case can be different.

tannewt, 2017-09-29:
@mrmcwethy Sounds like we'd just need to rename it.

I'm ok with us varying where CPython platforms also vary.

Thanks for looking into this!

mrmcwethy, 2017-09-30:
I believe fixing this issue consists of the following tasks:

1. #define MICROPY_PY_STRUCT (0) for atmel-samd and esp8266 ports 
2. implement struct module in shared-bindings and/or shared-module
3. repair the bundle library .py code that uses import ustruct
4. change docu where needed

mrmcwethy, 2017-09-30:
what branch should i make my changes for this issue .  I still plan on making my own branch, but I wondered what should be the source branch.
Note that i have made the changes to to shared-bindings and shared-module.  The atmel side of things build with error, but the esp8266 has many unknown compile errors.  I am assuming that i do not have the appropriate sources for esp8266 to build.

dhalbert, 2017-09-30:
@mrmcwethy Fork this repo into your own copy. Then checkout `2.x` and then make a branch off that. When you're done commit your changes and push to GitHub. When you look on GitHub you'll see you can create a pull request. You may need to choose the `adafruit/circuitpython` `2.x` branch from the dropdown. It might default to `master`.

mrmcwethy, 2017-10-01:
Thanks, Dan.  I got the esp8366 to build as well.  The resultant uf2 runs correctly on Metro and CPX.  
I have a question of project design.  I noticed that there is a separation of code between shared-bindings and shared-module.  Can you define what code should go in either and what should not be in either?  My first attempt was to move internal STATIC function into share-module, but i think i need to do more.

dhalbert, 2017-10-01:
@tannewt designed the `shared-bindings` / `shared-module` split. This [doc page](http://circuitpython.readthedocs.io/en/latest/docs/common_hal.html) describes it in more detail. Basically, the wrapper code that implements the Python API and calls down to the lower level HAL (hardware abstraction layer/level) is in `shared-bindings/`. Lower-level code that is architecture-independent and can be shared between ports lives in `shared-module/`. Then there are (for now) `atmel-samd/common-hal/` and `esp8266/common-hal/`, which implement HAL  code that is architecture-dependent.

mrmcwethy, 2017-10-01:
I have make changes to several bundle library projects: lis3dh, max31855, pca9685, rgb-display, si7020, and register.  which branch of these projects should I be updating.  I assume master, but I want to make sure first.  The changes are simple: import ustruct -> try: import struct except: import ustruct as struct.  change module references from ustruct. -> struct.

tannewt, 2017-10-02:
Master please @mrmcwethy. For small projects, master is the place to go.

tannewt, 2017-10-31:
Done in b41cbe9256d3f7105a70ad3d2e4fb64180f81e09

closed 2017-10-31 by tannewt