adafruit/circuitpython #213: trinket mo -- no SPI pins in board

deshipu, 2017-08-30:
Trying to run any of the SPI examples, I get:
```
>>> spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'SCK'
```

Changing this to:

```
spi = busio.SPI(clock=board.D3, MOSI=board.D4, MISO=board.D2)
```
makes it work, but then I have to write board-specific code.

tannewt, 2017-08-30:
How does it work when there is no chip select line? Isn't that usually used to signify transaction boundaries?

deshipu, 2017-08-30:
No idea, honestly, but it works. It would be nice to have those pin names in the board module, though.

tannewt, 2017-08-30:
I hesitate to add the pin names because most SPI devices won't work due to no pin for the chip select.

deshipu, 2017-08-30:
Any pin can be specified as chip select, and it's the driver for the particular device that is responsible for toggling it.

deshipu, 2017-08-30:
I really have no idea about how the chip select line is related to this issue at all.

tannewt, 2017-08-31:
I was confusing the Gemma which has 3 GPIO with the Trinket which has enough for SPI. Pull request proposed with the names.

commit 17ec02a7 (2017-08-31): atmel-samd: Add SPI pin names to the Trinket. Fixes #213 (#217)
closed 2017-08-31 by dhalbert