adafruit/circuitpython #215: Fix warnings in sphinx documentation build

dhalbert, 2017-08-31:
There are unconnected pages, ambiguous references, and warnings about the new `templates/replace.inc` file, which came in with MPy v1.9.2

willingc, 2017-08-31:
@dhalbert I noticed that too. I've been refactoring the docs since the build is a bit odd as it stands. Trying to restructure so the docs will be more maintainable over time. Feel free to tackle yourself or I'm happy to take this on.

dhalbert, 2017-08-31:
@willingc If you would like to that would be great! Thanks! It would also be nice to suppress the irrelevant stuff (e.g., `pyboard` module documentation).

tannewt, 2017-08-31:
Thanks for doing this! One thing I changed from MicroPython is defaulting the role to `any` this means more references are missing but some are connected and are more useful.

commit 20dd3b1e (2018-02-22): Delete a bunch of docs, drivers and examples not relevant to CircuitPython.

This fixes #345 and fixes #215.
closed 2018-02-28 by dhalbert