adafruit/circuitpython #219: Trinket M0: SPI not working correctly

deshipu, 2017-08-31:
I'm trying to get a ST7735 display to work with Trinket M0, and it seems that Trinket's SPI behaves very strangely. Here is a screenshot of a capture of a fragment of the transmission:

[image]

and here is a close-up on one of the fragments:

[image]

Here's the complete capture for analysis: 
[cap.logicdata.zip](https://github.com/adafruit/circuitpython/files/1268519/cap.logicdata.zip)

And here is the code that I used to capture that:
```
>>> import busio, digitalio, board
>>> from adafruit_rgb_display import st7735
>>> cs = digitalio.DigitalInOut(board.APA102_MOSI)
>>> dc = digitalio.DigitalInOut(board.D0)
>>> rst = digitalio.DigitalInOut(board.D1)
>>> spi = busio.SPI(clock=board.D3, MOSI=board.D4)
>>> d = st7735.ST7735R(spi, cs=cs, dc=dc, rst=rst)
```

Looking at the capture, two things are immediately striking:

1. the polarity of the data line is wrong
2. there are strange blips happening on both clock and data lines in between the actual bytes being transmitted

tannewt, 2017-08-31:
What version of CircuitPython are you using?

deshipu, 2017-08-31:
Sorry, channel 0 is the clock, channel 1 is the data (channel 2 is the D/C). Running CircuitPython 1.0.0 that came flashed on the Teensy M0.

deshipu, 2017-09-01:
I figured out where the "blips" are coming from — the BusDevice library is calling `spi.configure` before every transmission, and that toggles the clock and data lines.

tannewt, 2017-09-01:
Weird! What settings do you have? Configure shouldn't do anything if the setting are the same as before. Relevant code is here: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/common-hal/busio/SPI.c#L171

deshipu, 2017-09-01:
I didn't touch the default settings — the code I pasted in this bug is the only thing running since reset. I agree it's weird, especially since it only happens on the Trinket M0 — the Feather M0 Express is fine.

deshipu, 2017-09-01:
Actually, I don't see you setting the `self->current_baudrate` anywhere in that code?

PR #223 (2017-09-01): atmel-samd: Remember SPI baudrate.
This prevents the SERCOM from blipping the data line on each
transaction and therefore fixes #219.
commit e2f39a8d (2017-09-01): atmel-samd: Remember SPI baudrate.

This prevents the SERCOM from blipping the data line on each
transaction and therefore fixes #219.
closed 2017-09-01 by deshipu