adafruit/circuitpython #221: atmel-samd: os file operations that would write give error 30

dhalbert, 2017-08-31:
```
>>> import os
>>> os.mkdir("foo34")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 30
>>> os.rename("main.py", "foo.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 30
```
It would be better to give some indication that the filesystem was read-only instead of a numeric error.

tannewt, 2017-09-01:
Here is what CPython 3.6 does in the REPL and what we should mimic.

```
>>> import os
>>> os.mkdir("/Volumes/Untitled/test")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 30] Read-only file system: '/Volumes/Untitled/test'
```

Here are the relevant properties on it:
```
>>> error.errno
30
>>> error.strerror
'Read-only file system'
>>> error.filename
'/Volumes/Untitled/test'
```

dhalbert, 2017-09-01:
Revising title: on atmel-samd, all errors are reported only numerically. On esp8266 and unix, the errno name is given as well. As @tannewt shows above, both could be better.

```
>>> os.listdir("main.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 2
```

esp8266 and unix ports:
```
>>> os.listdir("boot.py")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```

Sigafoos, 2017-09-12:
This may be a separate issue, but `OSError` doesn't seem to have some of the attributes the CPython docs says it does:

> This exception is derived from EnvironmentError. It is raised when a function returns a system-related error (not for illegal argument types or other incidental errors). The errno attribute is a numeric error code from errno, and the strerror attribute is the corresponding string, as would be printed by the C function perror(). See the module errno, which contains names for the error codes defined by the underlying operating system.
> 
> For exceptions that involve a file system path (such as chdir() or unlink()), the exception instance will contain a third attribute, filename, which is the file name passed to the function.

```>>> v = None
>>> try:
...     fp = open("/x", "a")
... except OSError as e:
...     v = e
...
>>> v.args[0]
30
>>> v.errno
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'OSError' object has no attribute 'errno'
>>> v.strerror
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'OSError' object has no attribute 'strerror'```

process1183, 2017-10-20:
This is something I would like to work on.

godlygeek, 2018-05-14:
I'm going to take a swing at this one.