adafruit/circuitpython #226: Example for audiobusio.PDMIn doesn't work

ntoll, 2017-09-02:
When using the examples here: https://circuitpython.readthedocs.io/en/latest/shared-bindings/audiobusio/PDMIn.html I get the following error using RC1 of CircuitPython 2.0:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid clock pin
```

I notice that the argument order for instantiating the `PDMIn` class has changed (`MICROPHONE_CLOCK` used to come first). Can you confirm that the signature for `__init__` is not going to change further?

tannewt, 2017-09-03:
It looks like a doc mistake. Does it work with clock first? I'll fix it up on Tuesday after our long weekend.

ntoll, 2017-09-04:
OK... With clock first I just get a NotImplementedError when I try to instantiate the `PDWin` class. See the following REPL example:

```
Adafruit CircuitPython 2.0.0-rc.1 on 2017-09-01; Adafruit CircuitPlayground Express with samd21g18
>>> import audiobusio
>>> from board import *
>>> x = audiobusio.PDMIn(MICROPHONE_CLOCK, MICROPHONE_DATA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError:
```

However, if I try:

```
x = audiobusio.PDMIn(MICROPHONE_CLOCK, MICROPHONE_DATA, bit_depth=16, mono=True)
```

...it's successfully instantiated. The docs suggest that `bit_depth` and `mono` have the default values I use above (see https://github.com/adafruit/circuitpython/blob/master/atmel-samd/common-hal/audiobusio/PDMIn.c#L99 for the origin of the NotImplementedError).

Another niggle. If I just use a `bytearray` with the `record` method I get the following:

```
ValueError: destination buffer must be an array of type 'H' for bit_depth = 16
```

To fix this I've instantiated the buffer like this:

```
BUF = array.array('H', bytearray(8000))
```

But this doesn't feel very elegant. :-/

Hope this is useful. Thoughts?

tannewt, 2017-09-05:
Array is a bit weird yeah. Its a good way to have an explicit element type though.

I've got a change in the works that should turn on 8-bit support though it might suffer from not being very sensitive. It'll also correct the examples.

ntoll, 2017-09-05:
Great stuff. The important thing for me is that what ends up in the book is correct. So once there's some sort of stability in this regard then I can update the code examples to the correct API calls. ;-)

closed 2017-09-05 by tannewt

kevinjwalters, 2018-08-19:
On a similar theme, I just reported the example on https://learn.adafruit.com/sensor-plotting-with-mu-and-circuitpython/sound being broken, tested against a CPX board running 3.0.0. This looks like it's a change from 2.x, **frequency** is now **sample_rate** with no preservation of the old value for older code:

- https://circuitpython.readthedocs.io/en/2.x/shared-bindings/audiobusio/PDMIn.html
- https://circuitpython.readthedocs.io/en/latest/shared-bindings/audiobusio/PDMIn.html 

I'll mention @kattni as she's original author of that page.

PR #234 (2017-09-05): atmel-samd: Enable 8-bit audio recording support even though it'll be quiet.
Also update the examples.

Fixes #226
commit 6d9d6834 (2017-09-05): atmel-samd: Enable 8-bit audio recording support even though it'll be
quiet. Also update the examples.

Fixes #226