adafruit/circuitpython #229: i2c native doesnt require lock

ladyada, 2017-09-03:
shouldnt work :)

```
import board
import busio

BYTES_TO_READ = 2

with busio.I2C(board.SCL, board.SDA) as i2c:
    while not i2c.scan():
        pass
    print("I2C addresses found:", [hex(i) for i in i2c.scan()])

    i2c_addr = i2c.scan()[0]

    print("Reading each address")
    for addr in range(255):
        buffer = bytearray(BYTES_TO_READ)  # read n bytes from each addr
        try:
            i2c.writeto(i2c_addr, bytearray([addr]))
            i2c.readfrom_into(i2c_addr, buffer)
        except:
            # failed to read from that address, keep going
            continue
        print("\nAddress 0x%x : " % addr, end="")
        for b in buffer:
            print("0x%02x " % b, end="")
```

tannewt, 2017-09-06:
What board was this on? On a metro m0 express I get:

```
Adafruit CircuitPython 2.0.0-rc.1-dirty on 2017-09-05; Adafruit Metro M0 Express with samd21g18
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 24, in <module>
  File "code.py", line 8, in <module>
RuntimeError: Function requires lock.
```

ladyada, 2017-09-06:
im using 1.0 on trinket

tannewt, 2017-09-06:
I think that `check_lock` was being optimized away during LTO.

PR #240 (2017-09-06): shared-bindings: Prevent check_lock from being optimized away.
Fixes #229
commit 6467e3d4 (2017-09-06): shared-bindings: Prevent check_lock from being optimized away. (#240)

Fixes #229
closed 2017-09-06 by dhalbert