adafruit/circuitpython #232: Nuked main.py

ladyada, 2017-09-05:
of course i cant replicate it but while writing https://github.com/adafruit/Circuit_Pyrate i somehow REPL reset while the script was saving or ran before something synced and feather locked up, on reboot main.py was 0 bytes.

ladyada, 2017-09-06:
oooh figured it out

ok make a new main.py with:

`myname = input("type in a name: ")`

now in REPL, ^D or whatnot to run it. when it says type in a name, start typing a few chars BUT DONT HIT RETURN

now in main.py change the code and save file. REPL is hung, but USB MSC is still running. it doesnt zero out my main.py with this script but but it does with a longer script. i can tell you how to do that if you like

its a fairly nasty bug :(

tannewt, 2017-09-06:
Does this repro in 2.0? I know you've been doing most of your testing on 1.0. Brain is toast for tonight. Will look tomorrow.

willingc, 2017-09-06:
Confirming that this is also an issue with 2.0.0-beta1. Following the above process, it locked up my Terminal on the Mac. code.py was not zeroed out.
[image]

dhalbert, 2017-09-06:
Another clue or two:

I have a `main.py` that's just `input(":")`. I get it to run by typing ctrl-D at the repl. Then in a shell I do:
`$ touch /media/halbert/CIRCUITPY/main.py`
If I haven't typed anything at the ":" prompt, it soft-reboots fine.
If I have typed at least one character, it hangs and I have to hard-reset or unplug.

dhalbert, 2017-09-08:
Fixed by #241. That fixes handling of interrupts during stdin input, and this issue got fixed as a side-effect.

It does not fix main.py getting truncated, etc. That's the old delayed-write problem and is separate.

closed 2017-09-08 by dhalbert