adafruit/circuitpython #245: SD Cards producing OSError 19

tannewt, 2017-09-06:
```
Adafruit CircuitPython 2.0.0-rc.1 on 2017-09-01; Adafruit Feather M0 Adalogger with samd21g18
>>> import os
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> os.listdir()
['boot_out.txt', 'adafruit_sdcard.mpy', 'adafruit_bus_device', 'spi_device.mpy', 'i2c_device.mpy']
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.SD_CS)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
OSError: 19
>>> with open("/sd/test.txt", "w") as f:
... f.write("Hello world\n")
... 
... 
... 
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
OSError: 30
>>>
```

https://forums.adafruit.com/viewtopic.php?f=60&t=114744

hukuzatuna, 2017-09-07:
NOT replicated on Feather M0 Adalogger! The above code works without error (Class 10 Sandisk card).

tannewt, 2017-09-07:
This error can occur if the card isn't formatted. fabitencourt, does it work when connecting to a computer or in Arduino?

fabitencourt, 2017-09-08:
They are formatted fat32. Just saw the properties on my Windows 10 computer.

fabitencourt, 2017-09-08:
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 2.0.0-rc.1 on 2017-09-01; Adafruit Feather M0 Adalogger with samd21g18
>>> import os
>>> os.listdir()
['boot_out.txt', 'adafruit_sdcard.mpy', 'adafruit_bus_device', '.Trash-1000']
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.SD_CS)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "libraries/drivers/sdcard/adafruit_sdcard.py", line 84, in __init__
  File "libraries/drivers/sdcard/adafruit_sdcard.py", line 117, in _init_card
  File "libraries/drivers/sdcard/adafruit_sdcard.py", line 172, in _init_card_v2
OSError: timeout waiting for v2 card
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")
>>> with open("/sd/test.txt", "w") as f:
...     f.write("Hello world\n")
...
...
...
12
>>> os.listdir()
['boot_out.txt', 'adafruit_sdcard.mpy', 'adafruit_bus_device', '.Trash-1000']
>>> os.listdir('/sd')
['test.txt']
>>>

fabitencourt, 2017-09-08:
Adafruit CircuitPython 2.0.0-rc.1 on 2017-09-01; Adafruit Feather M0 Adalogger with samd21g18
>>> import os
>>> os.listdir()
['boot_out.txt', 'adafruit_sdcard.mpy', 'adafruit_bus_device', '.Trash-1000']
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.SD_CS)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")
>>> with open("/sd/test2.txt", "w") as f:
...     f.write("Salut tout le monde\n")
...     
...     
... 
20
>>> os.listdir('/sd')
['test.txt', 'test2.txt']
>>>

fabitencourt, 2017-09-08:
Just saw the blue 16G on gparted and it had a 4 M not allowed. I've formatted on fat 32. I'll try with that too.

fabitencourt, 2017-09-08:
No way. Neither of Lexar cards are working. The black 4G and the blue 16 G too. Both Lexar cards don't seems to work for now.

tannewt, 2017-09-08:
Did you get it working with a different card? I ended up with a Kodak card instead of Lexar but I'll give it a try today.

tannewt, 2017-09-08:
I believe I'm seeing the same thing with a Samsung EVO Select card. It works with my Mac via card reader but produces an OSError 19 in CircuitPython. Looking at it now.

fabitencourt, 2017-09-08:
I've downloaded the new version, candidate 2. I'll try it as soon as I can. I've noticed the DS3231 library is already there. Perfect. I'll give a try.

tannewt, 2017-09-08:
@fabitencourt can you please try this mpy (after unzipping)? Its for CircuitPython 2.x. What I found was that the Samsung cards I have don't like the chip select deinit and extra clocks between the read command response and the actual data.

The fix is in [this pull request](https://github.com/adafruit/Adafruit_CircuitPython_SD/pull/1).

[adafruit_sdcard.mpy.zip](https://github.com/adafruit/circuitpython/files/1289298/adafruit_sdcard.mpy.zip)

fabitencourt, 2017-09-08:
Sorry Scott, I can't do it right now. I'll be doing it by tomorrow morning, I'm at work now and I don't have my M0 Adalogger with me.

tannewt, 2017-09-08:
No rush! Thanks for the quick reply.

fabitencourt, 2017-09-08:
My pleasure.

jerryneedell, 2017-09-09:
@tannewt FYI - I am able to access a SanDisk 8GByte Class 10 card from my M0-Express via an Adalogger Featherwing. Using latest 2.0.0 and library bundle

tannewt, 2017-09-10:
Ok! Thanks for the report @fabitencourt. I will buy some more cards off of Amazon. I expected the newer library to work better. Perhaps they are even pickier about the chip select line. The arduino library is also much faster in talking with the chips.

fabitencourt, 2017-09-10:
In my opinion the new library adafruit_sdcard.mpy from bundle candidate 2
is the best I've used. But still doesn't work with my Lexar cards. But they
work very well with my sandisk 16G card. That's the one I'll use. Now I
will try the DS3231 library. That's essential to my logger project. It's
all like Christmas gifts all around....📦🍾🎁👍

cehiles, 2017-09-13:
I am experiencing similar behaviour with the following:
-Adafruit CircuitPython 2.0.0 on 2017-09-12; Adafruit Feather M0 Express with samd21g18
-Adafruit Adalogger FeatherWing
-adafruit-circuitpython-bundle-2.0.0-20170912
-SanDisk Ultra 64GB microSDXC UHS-I Card (Class 10)

>>> import os
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> os.listdir()
['boot_out.txt', '.Trash-1000', 'code.py', 'lib']
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.D10)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 19

Is anyone able to send the exact product description for an SD card that they have working with CircuitPython?  Preferably 32GB or bigger.  I'd like to order a card that is known to work.  Thanks!

fabitencourt, 2017-09-13:
In my opinion you're missing some files on circuitpy's directory. Personally mine is like this and is working with a 16 G Verbatim card...

os.listdir()
['boot_out.txt', 'adafruit_sdcard.mpy', 'adafruit_bus_device', '.Trash-1000']

cehiles, 2017-09-13:
The necessary libraries are in the /lib folder.  However, I reformatted Fat32 using gparted and that got things working.  Thank you!

fabitencourt, 2017-09-13:
My pleasure. Really glad to know it. That's exactly what happened in my case.

tannewt, 2017-09-14:
@cehiles glad that worked!

meichr, 2017-09-19:
Adafruit CircuitPython 2.0.0 on 2017-09-12; Adafruit Feather M0 Express with samd21g18
>>> import os
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> os.listdir()
['boot_out.txt', '.fseventsd', 'lib', '._lib', 'code.py', '._code.py', '.Trashes']
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.D10)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")

After the last line the terminal freezes, no reaction for at least 10 minutes.
Test of SD card in "Adalogger FeatherWing - RTC + SD Add-on".
Tested SD cards with this reaction:
Transcend 4GB class 10, Transcend 4GB class 2.

meichr, 2017-09-19:
Adafruit CircuitPython 2.0.0 on 2017-09-12; Adafruit Feather M0 Express with samd21g18
>>> import os
>>> import adafruit_sdcard
>>> import busio
>>> import digitalio
>>> import board
>>> import storage
>>> os.listdir()
['boot_out.txt', '.fseventsd', 'lib', '._lib', 'code.py', '._code.py', '.Trashes']
>>> spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = digitalio.DigitalInOut(board.D10)
>>> sdcard = adafruit_sdcard.SDCard(spi, cs)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 19
>>> with open("/sd/test.txt", "w") as f:
...     f.write("Hello world\n")
...     
...     
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 30
>>> 

Test of SD card in "Adalogger FeatherWing - RTC + SD Add-on".
Tested SD card with this reaction:
Samsung EVO 32GB UHS-I, cass 10 (up to 48 MB/s).
SD Card had been quick formatted with SDFormatter Version 4.0 (macOS Sierra).

tannewt, 2017-09-19:
Ok, I've ordered Transcend, Verbatim and Lexar cards to test with. @meichr could you post a picture of the cards? I think I've tested a similar Samsung card and it worked ok.

My test cards so far:

[image]

meichr, 2017-09-19:
@tannewt I've done the above tests with the library-2.0.0 bundle from Sep. 15th and thought that it already contained the above update from Sep 9th. My error!
The pictured SanDisk sd card works with the Sep. 15th version and the above Sep. 9th version. The Transcend sd cards and the Samsung sd card work with the above Sep. 9th version, but not with the bundled Sep. 15th version.
[image]

These cards work with the above Sep. 9th library file.

tannewt, 2017-09-19:
Wups, looks like I forgot to request a review from @dhalbert on it! The fix is here: https://github.com/adafruit/Adafruit_CircuitPython_SD/pull/1

meichr, 2017-09-19:
But I forgot to thank you for the fixes :-) .

closed 2017-09-27 by tdicola