adafruit/circuitpython #249: os.listdir inconsistent

tannewt, 2017-09-08:
After an SD card is mounted, `os.listdir` returns different results depending on if the path name is given or implicit through the current working directory.

```
print(os.listdir(os.getcwd()))
print(os.listdir())
```

Produces:
```
[b'sd', 'boot_out.txt', '.fseventsd', 'adafruit_sdcard.mpy', 'code.py', '._code.py', 'adafruit_bus_device']
['boot_out.txt', '.fseventsd', 'adafruit_sdcard.mpy', 'code.py', '._code.py', 'adafruit_bus_device']
```

RidleyLarsen, 2018-05-14:
I'm going to take a look at this.

RidleyLarsen, 2018-05-14:
This is a little outside my comfort zone (which is great!) but I'm reading through the C code. My first intuition is that when the listdir function has to infer the directory, it's not recognizing it as the root directory in the way it would when you explicitly provide it with the root directory (via `getcwd()`).

dhalbert, 2018-06-15:
It turns out the reason for this has to do with what's mounted where, and how MicroPython was set up to handle mounted filesystems. On the pyboard, the internal flash is mounted as `/flash`, and an SD card is mounted as `/sd`:
```
>>> os.listdir('/')
['flash', 'sd']
```
On the pyboard,  `/` is not a filesystem: it's only a location for mount points. On the pyboard, you cannot create files or directories in `/`:
```
>>> f = open("/foo.txt", 'w')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>> os.mkdir('/mnt')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```
The assumption that `/` is only for mount points is carried through in `mp_vfs_lookup_path()`, which is called indirectly by `os.listdir()`. In CircuitPython we mount the the on-board filesystem (`CIRCUITPY`) on `/`, not someplace like `/flash` or `/circuitpy`. So the fix is probably to make `mp_vfs_lookup_path` be more flexible.

deshipu, 2018-06-15:
I don't know if this belongs here, but we also had a user report that if you create a directory called `sd` and also mount the sdcard as sd, you get "sd" listed twice by listdir.

dhalbert, 2018-06-15:
@deshipu, hmm, that's another bug. It should not let you use a mount point that's the same name as an existing file or dir.

PR #932 (2018-06-15): fix os.listdir() when current dir is '/'
This makes `os.listdir()` include mounted filesystems when the current directory is `/`.

Also fixes the mounted filesystem sometimes showing up as, e.g., `b'sd'` instead of `'sd'`. Some code was treating a `char *` as an `mp_obj_t` by accident.

The handling of `""` by `mp_vfs_lookup_path()` is not ideal when the main filesystem is mounted at `/`. This avoids that by actually passing in the current directory instead of `""`.

Note that `os.listdir('.')` when the current directory is `/` still does not include mounted filesystems, because all the path l [... cut]
closed 2018-06-15 by tannewt