adafruit/circuitpython #250: Using a fona module with current networking stack

diogoviannaaraujo, 2017-09-08:
Hi,

Ive been experimenting with circuit python and one of yours feather m0 and as far as I can understand the networking stack is all made for wifi usage. Is it possible to write a interface to use it with an GSM module, like fona?

Then most of the already written code for the networking stack like mqtt and sockets can be used.

tannewt, 2017-09-08:
Hi @diogoviannaaraujo I have no idea. We've inherited the networking code from Micropython and haven't looked into it ourselves.

diogoviannaaraujo, 2017-09-08:
How do you guys implement socket on the ESP8266?

willingc, 2017-09-08:
Hi @diogoviannaaraujo, I recommend taking a look at the Micropython code. There are some [networking examples with sockets](https://github.com/micropython/micropython/tree/master/examples/network) in their repo. Hope this helps you.

KurticusMaximus, 2017-09-09:
@diogoviannaaraujo I've used socket exactly how it is in the networking guide for micropython and it works. I'm also researching using urequests. If you have any questions, I may be able to help.

diogoviannaaraujo, 2017-09-09:
@KurticusMaximus @willingc  I was asking how do they implement the networking  abstraction on the ESP8266 because it is too a "AT" based network stack. So I could try to implement a similar one to SIM808.

Ive done some more research and discovered that micro python implements the Network module witch implements most of the connection, configuring and socket abstraction in a NIC. But at the moment I couldn't find any Python based examples on creating a NIC Class.

Oh and I looks like they are 'per port', witch seems weird to me since I was expecting most of the "AT" commands abstractions would be the same on most platforms...

Ill probably try to find how to add a NIC Class to network module all in python, if thing go bad I may try to make it based on some of the C code in the ports, like:

https://github.com/micropython/micropython/blob/cc7fece309b0ce6d361cade8690b6c3a162d7378/ports/stm32/modnwwiznet5k.c

or

https://github.com/micropython/micropython/blob/4a93801c12898898744131f57d79cf216d0861b3/ports/esp8266/modnetwork.c

ladyada, 2018-07-25:
fyi there's some stuff here for 'DIY' cellular commands
https://github.com/HologramEducation/AT-Workshop

ladyada, 2020-04-19:
fona library is here, its very early but can be tested if folks want to try it out!
https://github.com/adafruit/Adafruit_CircuitPython_FONA

closed 2020-04-19 by ladyada