adafruit/circuitpython #251: Raw REPL crashing

eighthree, 2017-09-12:
It's possible for Raw REPL will stop working properly if `main.py` contains code that looks like what I whipped up below:

**main.py contents**
```
import time
from machine import ADC

x = True
while True:
  startMillis = time.monotonic()
  signalMax = 0
  signalMin = 1023
while time.monotonic() - startMillis < 10:
  adc = ADC(0)
  sample = adc.read()
if sample < 1024:
  if sample > signalMax:
      signalMax = sample
  elif sample < signalMin:
      signalMin = sample
peakToPeak = signalMax - signalMin
print(peakToPeak)
```

Normally, I'd get something like:
`ampy.pyboard.PyboardError: ('exception', '', 'Traceback (most recent call last):\r\n  File "<stdin>", line 14\r\nSyntaxError: invalid syntax\r\n')`
but running this `main.py` code through ampy causes terminal to hang.

**To reproduce**
1. ) Flash CP 2.0.0 rc2 on an ESP8266
2. ) Run `main.py` (hangs)
3. ) Put `main.py` (works ok!)
4. ) Reset/Unplug unit
5. ) Try running `main.py` through ampy (hangs) *
6. ) screen /dev/ttyUSB0 115200 (_see video for behaviour_)
7. ) Try running -or- putting file via ampy returns `File "/usr/local/lib/python2.7/dist-packages/ampy/pyboard.py", line 202, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl`

Observed on an **ESP8266** running CP 2.0.0-rc.2 / Ubuntu 16
Latest ampy & esptools

I've attached a screen captured video (webm format) demonstrating this issue:
[ESP Development Environment-2017-09-12T01-24-24-828588200Z.zip](https://github.com/adafruit/circuitpython/files/1294489/ESP.Development.Environment-2017-09-12T01-24-24-828588200Z.zip)

Here's the error when trying to use ampy when the Raw REPL fails:
```
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/ampy/cli.py", line 271, in run
    output = board_files.run(local_file, not no_output)
  File "/usr/local/lib/python2.7/dist-packages/ampy/files.py", line 221, in run
    self._pyboard.enter_raw_repl()
  File "/usr/local/lib/python2.7/dist-packages/ampy/pyboard.py", line 202, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

**What fixes the issue**
Re-flash the firmware.

jerryneedell, 2017-09-12:
Your first "while" loop will never exit.  It just executes the next 3 lines repeatedly. You need to revise your indentation.

jerryneedell, 2017-09-12:
something more like:
```
import time
from machine import ADC

x = True
while True:
  startMillis = time.monotonic()
  signalMax = 0
  signalMin = 1023
  while time.monotonic() - startMillis < 10:
    adc = ADC(0)
    sample = adc.read()
  if sample < 1024:
    if sample > signalMax:
      signalMax = sample
    elif sample < signalMin:
      signalMin = sample
  peakToPeak = signalMax - signalMin 
  print(peakToPeak)
```

jerryneedell, 2017-09-12:
RE your comment #6 , #7
You can't run ampy if you have a screen session open. Either ampy or screen can access the USB port at a time not at the same time

eighthree, 2017-09-12:
Thanks Jerry! Will revise the indentation.

To clarify No. 6 and No. 7: I'm not trying to have ampy and screen running at the same time. I could reset the unit , unplug it from power -- and Raw REPL will exhibit the same behaviour as shown in the video. 

No. 5 when it hangs, I abort the process (Ctrl + C) before opening a screen session.
`sudo ampy -- port /dev/ttyUSB0 run 'home/timothyDocuments/development/circuitpython/analog-read/main.py'
^C
Aborted!`

closed 2017-09-12 by tannewt