adafruit/circuitpython #252: 2.0.0 - bundle mpy doenst work, raw py does

ladyada, 2017-09-15:
code to test the internal dotstar on trinket (or gemma)

```
# Trinket Dotstar

from digitalio import *
import board
import adafruit_dotstar as dotstar
import time

numpix = 1
strip = dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1, brightness=0.2)

led = DigitalInOut(board.D13)
led.direction = Direction.OUTPUT

def wheel(pos):
    # Input a value 0 to 255 to get a color value.
    # The colours are a transition r - g - b - back to r.
    if (pos < 0):
        return [0, 0, 0]
    if (pos > 255):
        return [0, 0, 0]
    if (pos < 85):
        return [int(pos * 3), int(255 - (pos*3)), 0]
    elif (pos < 170):
        pos -= 85
        return [int(255 - pos*3), 0, int(pos*3)]
    else:
        pos -= 170
        return [0, int(pos*3), int(255 - pos*3)]

def rainbow_cycle(wait):
    for j in range(255):
        for i in range(len(strip)):
            idx = int ((i * 256 / len(strip)) + j)
            strip[i] = wheel(idx & 255)
        strip.show()
        time.sleep(wait)

while True:
    strip.fill([255, 0, 0])
    strip.show()
    time.sleep(1)

    strip.fill([0, 255, 0])
    strip.show()
    time.sleep(1)

    strip.fill([0, 0, 255])
    strip.show()
    time.sleep(1)

    rainbow_cycle(0)
```

if you are using https://raw.githubusercontent.com/adafruit/Adafruit_CircuitPython_DotStar/master/adafruit_dotstar.py it works as expected

using the bundle mpy does not but no error, just doenst make the LED do the right thing :)
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20170912
LED does flicker so something is happening?

dhalbert, 2017-09-15:
(diagnosed in chat)

Whoops! We updated Adafruit_CircuitPython_DotStar, but didn't publish a new bundle. So it was version skew. Bundle updated: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20170915

closed 2017-09-15 by dhalbert

dglaude, 2017-09-16:
This plus the fact that the name changed make the demo code distributed in
Gemma M0 and Trinket M0 not work when upgrading from 1.0.0 to 2.0.0.
And since the CircuitPython filesystem survive the upgrade, the hardware
appear to be not working (the led not blinking the right way).
Please provide to early adopter (those with hardware loaded with 1.0.0 and
demo code) a zip file that will match the content of new customer that will
get it with 2.0.0.
Both Gemma and Trinket have no space to load the full bundle, so the
minimal set you plan to distribute for demo would be great to have.

tannewt, 2017-09-16:
The contents of the default file system are available in the learn guide: https://learn.adafruit.com/adafruit-gemma-m0/installing-libraries?view=all#downloads https://learn.adafruit.com/adafruit-trinket-m0-circuitpython-arduino/downloads?view=all#downloads

Its still the 1.0 contents for now because the manufacturing hasn't been switched fully to 2.0 yet.

To upgrade your existing installation you'll need to copy each library in the lib directory from the bundle and replace the old version. Then in the code.py replace `dotstar` with `adafruit_dotstar` and swap the DotStar initialization SCK and MOSI so SCK is first. This should be the last major API change for DotStar that was done to get it consistent with other libraries. You shouldn't need to update it in the future.

ladyada, 2017-09-16:
yah we just did this yesterday at 7pm :) will update the bundle shortly!

ladyada, 2017-09-17:
hiya the files have been updated!