adafruit/circuitpython #253: 2.0.0 - do we undo the mcu temperature ADC changeover?

ladyada, 2017-09-15:
i think after mcu temperature is read the MUX isnt cleared or something because analog reads have a max range change?

see demo, try commenting out mcu temp print out, note analog voltage changes. with temp read, voltage never gets above 0.21 (4096 raw). using trinket but temp read is correct!

```
import board
from analogio import AnalogOut, AnalogIn
import time
import microcontroller

# Analog input on D0
analog1in = AnalogIn(board.D0)

######################### HELPERS ##############################

# Helper to convert analog input to voltage
def getVoltage(pin):
    return (pin.value * 3.3) / 65536

######################### MAIN LOOP ##############################

while True:
  #print("\nTemp: %0.1f *C" % microcontroller.cpu.temperature, end="")

  # Read analog voltage on D0
  print("\tD0: %0.2f V" % getVoltage(analog1in), end="")

  time.sleep(0.01) # make bigger to slow down
  print("")
```

dhalbert, 2017-09-15:
We setup and teardown the ADC struct each time. However, the doc says to discard the first reading. The temp code does that but the regular AnalogIn code does not. I'll test more thoroughly but if you are still setup to test try repeating the getVoltage call a second time.

ladyada, 2017-09-15:
ooh ok will do!

dhalbert, 2017-09-15:
Trying this in the REPL and I get ~4095 all the time as you do. The ASF calls we make are supposed to reset all the ADC logic completely, but apparently not. I'll look further. The double read is just because the ADC supposedly doesn't stabilize after you change the voltage reference: http://www.atmel.com/images/Atmel-42645-ADC-Configurations-with-Examples_ApplicationNote_AT11481.pdf, page 28:
>Discard the first conversion result whenever there is a change in ADC configuration like voltage
reference / ADC channel change

PR #255 (2017-09-16): Re-initialize ADC before every AnalogIn read.
`microcontroller.cpu.temperature` uses different ADC settings, and caused
AnalogIn to give wrong answers. AnalogIn can no longer assume it's the
only user of the ADC.

This fixes #253, at the expense of re-initing the ADC on every read, and also doing double reads to ensure accuracy of the result.

A more sophisticated implementation would remember the last ADC settings and not re-init if they were the same. If we're not going to do that, the AnalogIn code could be simplified to not save the ADC structs on the heap, but simply re-create them each t [... cut]
closed 2017-09-16 by dhalbert