adafruit/circuitpython #256: trinket_m0 pins.c file does not define A1,A2,A3,A4

jerryneedell, 2017-09-16:
The guide https://learn.adafruit.com/adafruit-trinket-m0-circuitpython-arduino/downloads?view=all
describes the us of mnemonic A1,A2,A3,A4 but they are not defined in the pins.c file.
Using the correct D# works, but I wondered if this was an intentional omission.
The mapping is bit confusing since Ax does not got to Dx in all cases.

tannewt, 2017-09-18:
Nope, I don't think its intentional. Mind taking a stab at adding them? Thanks!

jerryneedell, 2017-09-18:
I'll be glad to. Once cancer, and I can check this when I add them, is does this add to the code size? Since the trinket only has ~40bytes to spare, it may be better to deal with this via documentation rather than adding Aliases. Just a thought - I'll give it a try in the next few days.

tannewt, 2017-09-18:
Yeah, it will add a bit of size. Its worth a shot. The binary size will
change a bunch with 3.0 and ASF4 anyway. Thanks!

dhalbert, 2017-09-18:
You can gain space in the non-Express builds by adjusting `-finline-limit=<nn>` in `atmel-samd/Makefile`. Lower it by 1 or few at a time. It controls how big inlined routines can be. The value was recently to make room to add some code to dim the status pixel, and now there are 256 bytes free.

jerryneedell, 2017-09-18:
Great! I'll give that a try as well. The 40 bytes reference was to the 2 .0.0 I did not note the size on the latest. I'll be more attentive!

jerryneedell, 2017-09-20:
Submitted pull request. I hope I did it correctly. Let me know if I needed to do something else.

jerryneedell, 2017-09-20:
I did not mess with makefile. Still 187 bytes free!

tannewt, 2017-09-25:
Fixed in 3ad01ddb04b160b74b4baee9fce043da17dbcef5

closed 2017-09-25 by tannewt