adafruit/circuitpython #268: Add `touchio` support to the M4.

tannewt, 2017-09-19:


tannewt, 2018-05-22:
We're not going to block 3.0 on this. We'll add support once https://github.com/adafruit/Adafruit_FreeTouch supports it.

kenidowa, 2019-03-08:
Will this ever happen?

ladyada, 2019-03-09:
@kenidowa wanna help? how are you at reverse engineering ARM assembly :)

siddacious, 2019-03-09:
I've been seriously considering giving this a try. I have no idea how successful I'll be but I may or may not have watched your qtouch videos more than a few times and sought out an alternative to IDA.

Good money says I'll fail miserably but maybe I'd at least be able to tell someone more  what _not_ to try  :)

ladyada, 2019-03-09:
lol could look at the new free ghidra?

kenidowa, 2019-03-10:
So, I asked about Touchio because I found it really usefully for standalone application of a TrinketMO but when I tried using it on ItsyBitsyMO on larger applications I ran into code space issues.  So, moved to ItsyBitsyM4 but then found it doesn't have Touchio.  So, after watching 2yr. old video of yours on it.  Is it that the SAMD51 touch is so different from SAMD21 or is it the amount of code space it requires in the base that's the problem.  Looking at data sheets, it isn't apparent to me that they should be that different.

ladyada, 2019-03-10:
yep, its completely different

dhalbert, 2019-05-10:
Alternatively, we will do this via #1517, adapting the code in #1499.

PR #2071 (2019-08-22): make nrf touchio be generic: now available for SAMD51 too
- Generalize the nRF "charge up pin" touchio implementation so it can be used by all ports, and move it to `shared-module/touchio`.
- Turn on generic `touchio` for SAMD51. Now all CircuitPython ports have `touchio`.
- Add `CIRCUITPY_TOUCHIO_USE_NATIVE` flag to indicate whether to use `common-hal/touchio` or `shared-module/touchio`. Because SAMD21 has a native implementation, but SAMD51 does not, it was difficult to do this automatically, using, say, weak references. The lists of source directories (`SRC_COMMON_HAL` and `SRC_SHARED_MODUL [... cut]
closed 2019-08-22 by tannewt