adafruit/circuitpython #269: Have drivers and docs for all FeatherWings and breakouts.

tannewt, 2017-09-19:
First, lets track all FeatherWings and breakouts in this [doc](https://docs.google.com/spreadsheets/d/1vrxxjPMWX8WxU8E0C34_fWaERUAXAC9RC157LBjHv8M/edit?usp=sharing). Here are what the columns are:

## FeatherWings

* FeatherWing Name - Name from the Adafruit store
* Store Link - Link to the product page
* Learn Guide - Link to the product's learn guide
* CircuitPython Driver 1 - Link to one related CircuitPython driver
* CircuitPython Driver 2 - Link to another related CircuitPython driver
* Drivers in Bundle? - **Yes** when the driver is included in the [Adafruit bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle).
* Drivers have ReadTheDocs? - **Yes** when the driver has its own API docs on ReadTheDocs.
* ReadTheDocs linked from main docs? - **Yes** when the driver's docs are linked to from the main docs. ([here](https://circuitpython.readthedocs.io/en/latest/docs/drivers.html) for now.)

## Breakouts

* Breakout Name - Name from the Adafruit store
* Store Link - Link to the product page
* Learn Guide - Link to the product's learn guide
* CircuitPython Driver - Link to one related CircuitPython driver
* Drivers in Bundle? - **Yes** when the driver is included in the [Adafruit bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle).
* Drivers have ReadTheDocs? - **Yes** when the driver has its own API docs on ReadTheDocs.
* ReadTheDocs linked from main docs? - **Yes** when the driver's docs are linked to from the main docs. ([here](https://circuitpython.readthedocs.io/en/latest/docs/drivers.html) for now.)

Once we identify holes in our coverage we can make separate issues to track filling them. Thanks!

deshipu, 2017-09-23:
I updated this to mark "4-Digit 7-Segment LED Matrix Display FeatherWing" as supported, since the HT16K33 dirver handles it.

deshipu, 2017-09-23:
Seems like we only have radio, ethernet, GPS and music featherwings missing. I have the radio and music featherwings, but they are a bit complex, so I haven't written drivers for them yet. I also have the GPS module, but since ESP8266 only has one UART that is already used by the REPL, I didn't wite a driver for it.

There is a driver for the motor shield for micropython: https://github.com/mwm/Micropython-Drivers/blob/master/motors/AFMotorShield.py

tannewt, 2017-10-02:
I just added a column to both sheets to hold tracking issues that are used to note who is working on what.

tannewt, 2019-08-23:
We've been doing this and can formalize with AdaBot if we like.

closed 2019-08-23 by tannewt