adafruit/circuitpython #271: Help answer questions!

tannewt, 2017-09-19:
Although you consider yourself a CircuitPython beginner, its likely you already know something that other folks do not! Please share that knowledge!

There are a few places where people post questions and could use a helpful answer.

* [Adafruit CircuitPython Forums](https://forums.adafruit.com/viewforum.php?f=60)
* [Adafruit Discord](https://adafru.it/discord)
* [GitHub issues](https://github.com/adafruit/circuitpython/issues)

If cannot answer a question yourself and its been a few days since its been asked, then please join [Discord](https://adafru.it/discord) and mention `@circuitpython helpers` with a link to the question so someone else can help answer it.

larsks, 2018-05-02:
Is there a difference in use or audience between the forums and the discord instance?

jerryneedell, 2018-05-02:
In my experience, discord is far more interactive and depending on time of day there are several people willing and able to help.