adafruit/circuitpython #273: atmel-samd: Erratic I2C scanning behavior with nothing connected

tdicola, 2017-09-20:
I noticed an interesting thing on a feather M0 express with nothing attached, if you scan for I2C devices you actually get random results.  It's not necessarily a bug in CircuitPython but something worth noting that might trip folks up.  Here's an example of scanning with nothing connected:
```
>>> import busio
>>> import board
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> while not i2c.try_lock():
...     pass
... 
>>> i2c.scan()
[65, 102]
>>> i2c.scan()
[33, 71]
>>> i2c.scan()
[]
>>> i2c.scan()
[65, 102]
>>> i2c.scan()
[]
```
Likely this is from the SCL and SDA pins not having internal pull-ups applied and floating to random high/low values.  I wonder would it make sense to automatically turn these pull-ups on when I2C is initialized or even the board resets?

Impact-wise this isn't a huge issue.  The only trouble it might cause is with drivers that validate the device exists by scanning for them first.  In super rare cases they might get a false positive with nothing connected and interpret random noise as the expected I2C scan result.  The driver will eventually fail when it tries to talk to the device, but it might be nice to catch it earlier if possible.

As far as I2C behavior goes it's all good once you plug in a device (which has its own pull-ups), I get a steady and expected I2C scan result:
```
>>> i2c.scan()
[116]
>>> i2c.scan()
[116]
>>> i2c.scan()
[116]
```

tannewt, 2017-09-20:
Good point @tdicola. We could also fix it by scanning "found" addresses a second or third time before adding them to the list.

deshipu, 2017-09-23:
The problem with turning the pull-ups on is that then it makes it impossible for the I²C bus to talk to devices that use different voltages than 3.3V. Normally the pins are open-drain to make that possible. There can also be some bad interactions with pull-up resistors already included on many modules and shields — they would all add up, at some point making it impossible to pull the pin down. Granted, that particular problem is also present without the internal pull-ups, but it's made worse with them.

On the other hand, communicating with devices that use different voltages seems like an advanced topic, and requires some careful design of the circuit anyways, so perhaps the pull-ups could be enabled by default, but have an option for disabling them in the I²C initialization?

tdicola, 2017-10-11:
Some reference info, to compare with Arduino's I2C behavior it looks like there's no scanning built-in but if you use a scan sketch (like this one: https://playground.arduino.cc/Main/I2cScanner) it will require initializing the I2C library (Wire.begin() call) which turns on the AVR's pull-ups for SCL and SDA: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/Wire/src/utility/twi.c#L75-L77  It seems reasonable to match Arduino's behavior here and temporarily turn on pull-ups if they aren't already on when doing a scan.

tdicola, 2017-10-11:
And for the Pi it looks like they build in physical pull-ups (1.8K on the GPIO2 and GPIO3 lines for SCL, SDA): https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/Raspberry-Pi-3B-V1.2-Schematics.pdf  Seems reasonable to match the behavior and turn on pull-ups when doing an I2C action like scanning.

deshipu, 2017-10-11:
The fact that Arduino turns on pull-ups automatically is a well-known problem in the Arduino community, and the reason why the advanced users use other libraries (or just ad-hoc code) for I²C when the need to interact with mixed-voltage devices.

deshipu, 2017-10-11:
You also need to consider how the pull-ups will interact with the external resistors already included on the I²C lines. If high-speed communication is needed, strong pull-ups will be used, and that can be too strong fro the pins to drive when combined with the internal pull-ups.

dhalbert, 2017-10-11:
@deshipu Is it a problem electrically if we just turn on the pullups during `scan()` and then turn them off again?

deshipu, 2017-10-11:
I agree pull-ups are a reasonable default, but I would also strongly advise to have an option to suppress them.

deshipu, 2017-10-11:
> @deshipu Is it a problem electrically if we just turn on the pullups during scan() and then turn them off again?

Yes, it's still a problem. If you have a, say, 5V device on your I²C bus, and consequently, external pull-ups to 5V, then enabling the internal pull-ups to 3.3V will result in a short — through two pairs of large resistors, so maybe not that catastrophic, but a short none the less.

dhalbert, 2017-10-11:
If I understand correctly, Adafruit's philosophy is to have pullups on the I2C peripheral boards, not on the master board. So our core I2C implementations do not turn on internal pullups.

deshipu, 2017-10-11:
Well, Arduino does.

tannewt, 2017-10-11:
Why can't we fix this by double or triple probing addresses that are available? If its really just line noise from no pull-ups then it shouldn't be consistent over repeated probes. There is no need to turn on internal pull ups.

deshipu, 2017-10-11:
Alternatively, we could read a value on those pins before doing anything else, and throw an error about missing pull-ups if it's not consistently high.

tdicola, 2017-10-11:
Let's not lose the forest between the trees.  The original issue here is that I2C scanning is erratic and weird--if you're at the REPL and learning about I2C it's strange to have nothing connected but see random addresses returned.  It's good to get the positive feedback that with nothing connected no devices appear in a scan, then connect a device and see it's now available.  This is the same behavior in Linux with the Pi (call i2c-detect with nothing attached and you see nothing because of built-in pull-ups to the Pi's board), and the same behavior with Arduino.  Turning on the pull-ups during a scan seems like a reasonable approach that won't impact more advanced uses (sharing different voltages).

I'm not a fan of multiple probes, it will increase the time to scan and is still potentially flakey.  Would you delay a bit between scans and 'average' the result?  Even then some interference could cause problems.

IMHO flip on pull-ups for scan, it will make the most sense for the most users.

Andon-A, 2017-10-11:
My opinion, as a generic basic user, is default enable but with option to override. As has been mentioned previously, the biggest conflicts are when talking with other-voltage devices, something that newer users generally shouldn't be worried about, and advanced users should know. It'd be fairly simple, I think, to put in a guide "If you're using this with a 3.3v board, make sure to turn off the default internal pullup resistors"

deshipu, 2017-10-11:
> IMHO flip on pull-ups for scan, it will make the most sense for the most users.
But you can damage your devices by doing a scan then if your bus is pulled up to anything else than the board's voltage. We don't want to damage devices by doing something as innocent as a scan.

tannewt, 2017-10-11:
Turning on pull ups risks changing the properties of the bus and causing other read failures.

I wouldn't multiple probe all addresses, I'd only do it on "found" addresses to confirm they are really there. The example at the top only has one phantom address so it only result in a few extra probes to confirm its there.

tdicola, 2017-10-11:
@deshipu That's a similar problem with the Pi and Arduino and they haven't had any issues.  IMHO it's an edge case to alert folks in docs and not worry about.  If you're mixing I2C voltages you're doing pretty advanced stuff and are knee deep into the inner workings.

deshipu, 2017-10-11:
On a different, but related, topic, I wonder how does it work when you have multiple Adafruit modules connected on the same I²C bus, and all the pull-up resistors on those modules stack up, potentially creating a pull-up that is too strong to drive low. Have there been any experiment about how many Adafruit modules you can connect at once before that happens?

tannewt, 2017-10-11:
@deshipu Lets not discuss that here.

tdicola, 2017-10-11:
@tannewt Not sure I follow the issue with properties of the bus?  This wouldn't be in in the I2C init but only in the scan.  Perhaps in the probe call it would check if pullups were previously enabled and turn them on and then off if not.

deshipu, 2017-10-11:
Sorry

deshipu, 2017-10-11:
@tdicola I think that if we are going to enable the pull-ups by default, we should do it in the init, not in the scan. Otherwise people will be surprised that their devices don't work even though they appear in the scan. I would propose to simply add a keyword parameter to the I2C constructor that suppresses the internal pullups when set to true.

tdicola, 2017-10-11:
Yeah I was thinking about that but I think it's better raised as a separate issue.  To me it seems like another odd edge case, you plugged in a device and are making I2C read & write calls but forgot to call init.  In that case no matter what your code won't work since the sercom, etc. initialization was never called.  The only other case is if you're using a device with no built-in pull-ups, but that's not typical for adafruit and most arduino add-ons.  It might be worth discussing but I think it's better as a separate bug.  Just keep this one within the scope of 'fix scanning so you don't get random addresses with nothing connected'.

deshipu, 2017-10-11:
I think that if you just want to fix the scanning, then checking if the pins are really consistently high before the scan is the best option, because:

1. it's completely passive, you are not sending any voltages down the wire,
2. it lets you throw a very specific error with a message explaining the issue,
3. it doesn't slow down the scan itself, just adds a constant small delay in front,
4. an explicit error is less confusing than just an empty list,
5. it will also handle the case where you connected something to those pins that shouldn't be there.

tdicola, 2017-10-11:
I think back to the original issue though, person at REPL learning about I2C and calling scan with nothing connected.  If they get an error vs. an empty list (which is how the Pi, Arduino, etc. all behave today) it seems weird to me.  Did they do something wrong?  Not really.  If the proposed behavior of turning on pull-ups has been used by Arduino & Pi users (millions and millions of people) it seems safe to apply here too.

deshipu, 2017-10-11:
I think that an error like "Pin XXX is not high. Is your device connected?" would be informative enough both for professionals and beginners. It's really not obvious to me that a wrong connection would result in an empty list of devices. The fact that both Arduino and RPi do that doesn't mean we can't do better.

And if you really think that an empty list is better, we can have an empty list returned instead of the error after this check. No need for the pull-ups.

deshipu, 2017-10-11:
And error would be better also because in the case when the device is connected (so the pull-ups are present) but broken or the SDA and SCL are swapped, you get a different result — an empty list. Then we can cover it in troubleshooting instructions.

Andon-A, 2017-10-11:
@tdicola I disagree with the sentiment about errors. If everything is hidden, then it can be really annoying and frustrating to figure out what's wrong with my device.

Errors can be phrased nicely - IE, "No pullup resistors detected. Is everything connected correctly?" lets people know precisely what is wrong without the 'scary' aspect of "Error 254150: Pin is floating" or different.

deshipu, 2017-10-11:
How about returning `None` in one case and empty list in the other then? We still give users the information, but no error?

I think that "errors are scary" approach is what made PHP as horrible language as it is — it tries very hard to not raise an error, which in turns leads to errors that go unnoticed.

Andon-A, 2017-10-11:
Errors shouldn't be scary. They should be informative and useful. By including informative and useful errors, the scariness of errors can be eroded

ladyada, 2018-04-03:
fixed with https://github.com/adafruit/circuitpython/commit/cebcec5e63d594dda93917ffbecfde6d6fe5c5ca 

works great, lets you know if you don't have pullups detected, available in 3.0 :)

closed 2018-04-03 by ladyada