adafruit/circuitpython #275: atmel-samd: PDMIn record called twice fails with pin already in use error

tdicola, 2017-09-21:
On CircuitPython 2.0 with Circuit Playground Express unfortunately I'm running into a lot of trouble getting mic samples.  If I use the record function twice (in separate context managers) I get a pin already in use error.  Here's a simple repo:

```
>>> import audiobusio
>>> import board
>>> with audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA) as mic:
...     samples = bytearray(160)
...     mic.record(samples, len(samples))
... 
>>> samples[0]
128
>>> with audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA) as mic:
...     samples = bytearray(160)
...     mic.record(samples, len(samples))
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin PA10 in use
```

The context manager surrounding the PDMIn creation should presumably be deinitializing the pins.  Perhaps something is up with the deinit logic?

dhalbert, 2017-09-22:
`deinit()` was just missing calls to `reset_pin()`.

dhalbert, 2017-10-03:
Fixed by #304.

commit b2dcc5bb (2017-09-29): reset pins on PDMIn deinit(). Fixes #275.
closed 2017-10-03 by dhalbert