adafruit/circuitpython #276: atmel-samd: PDMIn class locks up entire board when reused

tdicola, 2017-09-21:
Here's another interesting audio in bug, if I create a PDMIn and save it and try to use it twice to record (all within the context of a context manager for init and deinit) it locks up the entire board and it becomes unresponsive to Ctrl-C, Ctrl-D, etc.  Even worse if you put code like this in main.py you're in a bad state that isn't recoverable without erasing the SPI flash in Arduino.  Here's a simple repo:

```
Adafruit CircuitPython 2.0.0 on 2017-09-12; Adafruit CircuitPlayground Express with samd21g18
>>> import board
>>> import audiobusio
>>> mic = audiobusio.PDMIn(board.MICROPHONE_CLOCK, board.MICROPHONE_DATA)
>>> with mic as m:
...     samples = bytearray(160)
...     m.record(samples, len(samples))
... 
>>> samples[0]
128
>>> with mic as m:
...     samples = bytearray(160)
...     m.record(samples, len(samples))
... 
```
At this point the board is unresponsive and needs to be hard reset (don't put the code wasn't put in main.py!).

dhalbert, 2017-09-22:
@tdicola @tannewt I did reproduce this, and it's not fixed by my fix to #275. However, think there might be a larger issue here. I have a question here about the intention of the context managers and what can be done with `deinit()`'d objects.

There is no `__enter()__` implemented for these kinds of `io` objects. The pins are claimed in `__init()__`. `__exit__()` calls `deinit()` and releases the pins. So then you can't reuse the object again, because the pins aren't claimed any more and there's no way to get them back.

I just to want make sure I'm understanding that. Tony's sample code above shouldn't work even without the bug, right? I tried something similar with `analogio.AnalogIn`, which also does something bad when a `deinit()`'d object is reused. It crashes instead of hanging. I did:

```
import board
import analogio
ai = analogio.AnalogIn(board.A1)
ai.value   # prints a value
ai.deinit()
ai.value   # CRASH! My repl closed.
```

Seems like `io` objects should know when they're `deinit()`'d and refuse to do anything. I also tried `AnalogOut`, which doesn't crash: I can do `some_analagout.value = 3`, but it should probably complain that nothing is happening.

dhalbert, 2017-10-03:
Fixed by #304.

closed 2017-10-03 by dhalbert