adafruit/circuitpython #279: Add a C module for handling buttons.

deshipu, 2017-09-23:
Handling buttons is a very common task for microcontroller boards, but doing it properly is a surprisingly complex and advanced topic. We need interrupts, we need timers, we need software de-bouncing, etc. Instead of exposing all this complexity to the users, and letting them write the button-handling code themselves, I think it could make sense to have a `buttons` module built into CircuitPython, which could be configured to handle buttons connected to the GPIO pins (and built into some of the boards), perhaps similar to what Micro:bit has: http://microbit-micropython.readthedocs.io/en/latest/button.html

We would initialize the module by providing a list of pins, and then they would be automatically set to input with pull-ups, and an interrupt handler would be installed, which would handle the key presses and do de-bouncing. We could then call a function such as `get_keys()`, that would give us a set of keys that were pressed since the last call, and clear the buffer. Alternatively, a more complex API with Button objects and `was_pressed()` methods could be devised.

deshipu, 2017-09-29:
I have started on a proof-of-concept module here: https://github.com/pewpew-game/circuitpython/tree/buttons

Turns out we don't need the interrupts, since we need a timer for de-bouncing anyways, we can poll in that timer. For now I put the functions in the `samd` module, because right now the code is specific to that platform. Of course ultimately it would go to its own module.

There are two functions: ``setup_buttons`` takes up to eight DigitalInOut objects, and configures CircuitPython to scan them for button presses. ``get_buttons`` returns an int with the bits corresponding to the buttons that have been pressed set. It also clears the buffer, so the next time it is called, it will only return the buttons that have been pressed since the last call.

tannewt, 2017-09-29:
Why are you doing them all together rather than as individual objects?

I agree a new module would be better. Its OK to have ones that are only implemented on one platform to start with.

deshipu, 2017-09-29:
> Why are you doing them all together rather than as individual objects?

Memory and speed. And the fact that in practice you always want to check them all at once anyways, and have some big switch statement for the different combinations in your main loop.

deshipu, 2017-09-29:
I moved everything to its own module, and added some docstrings.
I think there should also be a compiler flag for disabling this, but not sure how to best do it.

tannewt, 2017-09-29:
Have you tried it separately? I think the API would make much more sense with individual button objects.

```
import button
import digitalio
import board
d = digitalio.DigitalInOut(board.D0)
b = button.Button(d)
while True:
    if b.pressed:
        print("pressed!")
```

How much speed is actually gained by this? Have you tried doing it in Python? I'm not convinced this is necessary.

deshipu, 2017-09-29:
That's not how you would use it in a usual program, though. Whether it's a menu, a game or a remote control device, you would have a main loop where you check all of the buttons at once on every frame, and decide what to do based on which buttons are pressed. Just think of a simple "press any key" check, to pick an extreme case. With your API, that would lead to a lot of boilerplate — either copy-pasted lines of code, or loops and lists everywhere. And of course all those objects consume extra memory, which we really don't have in excess.

deshipu, 2017-09-29:
You can take a look at two games in which I'm using a very similar API:
https://github.com/deshipu/pewpew/blob/master/games/snake.py
https://github.com/deshipu/pewpew/blob/master/games/tetris.py
It seems to be working rather well.

tannewt, 2017-09-29:
Thanks for the example links!

I was getting a bit distracted by the API. I want to focus on the need for a C module to do this instead. In my mind there are two main reasons to require a C module:

1. Functionality is only available through a C API.
2. Multiple steps must occur in a strictly timed sequence.

I don't believe 1 applies here because pins can be read through `digitalio.DigitalInOut` already and debounced in Python.

I'm also not sure 2 does either. The time between reading inputs is not critical because the results are persisted over ticks and read less frequently. Have you implemented `pew.keys()` in Python to compare it to the C equivalent? How much faster is the C version?

I don't think memory is a reason to create a C module because using a frozen module can reduce the heap impact of Python code.

deshipu, 2017-09-29:
I have tried a naive implementation of `pew.keys()` in Python, of course. The games were completely unplayable because button presses would be missed or registered several times.

The problem is that we can't poll the keys fast enough and still let the user code to run at the same time. 

The basic feature required here is the ability to poll the keys at regular intervals (so that de-bouncing is consistent) and fast enough (so that we don't miss short button presses) while at the same time letting the user code run normally, call blocking functions and wait on delays.

Note that I don't need this feature merged for my game console project, because I will be using a custom compiled firmware with it anyways. I just think that this could be generally useful in a range of use cases involving button-based user interfaces.

tannewt, 2017-09-29:
Ok, that makes total sense to me! That explanation should go in the module docs please. I'm going to move feedback to the PR from now.

deshipu, 2017-09-30:
I will try again, but I really wonder if it's worth it to fight for those 8 bytes, especially sice the global pointer to the object alone will take 4 bytes already, not even counting all the extra code that is needed.

tannewt, 2017-10-09:
This was done with the introduction of `gamepad` in f4981677b0b73d286f3af098a4111f8ba6d286bc.

closed 2017-10-09 by tannewt

hybotics, 2018-11-03:
I have worked with interrupts in Micropython. This implementation works very well. I would very much like to see this added to CircuitPython.

braddo99, 2019-11-29:
It would be great if the same interrupt style behavior was available for potentiometers as well. Not sure but, in keeping with the "gamepad" theme, maybe joysticks would require similar high performance analog input scanning?

deshipu, 2019-11-29:
@braddo99 I don't see why you would need this for a potentiometer? Just read the value when you need it. There is no debouncing necessary, and no "events" to remember — everything is smooth, so you can poll at the speed that is required by your game.

Do you have a specific use case? How would that work exactly?

PR #295 (2017-10-03): Add a C module for handling buttons.
Fix #279 
This adds a `buttons` module to the samd port, with two functions useful for button handling: `setup()` and `get_pressed()`. The buttons are automatically scanned every 32 ticks, de-bounced, and reported to the user as a bit mask.