adafruit/circuitpython #281: audioio crashes on Trinket M0

deshipu, 2017-09-24:
This is a bit of a self-inflicted wound, as the `audioio` module is not normally enabled on the Trinket M0, so this is obviously very low priority. But I really wanted to try it, so I compiled the `2.x` branch of the firmare with `touchio`, `analogio`, `storage` and `neopixel_write` disabled, and with `audioio` enabled. Here's the change I made to `mpconfigport.h`:

```
diff --git a/atmel-samd/mpconfigport.h b/atmel-samd/mpconfigport.h
index e99d190be..ff480f39d 100644
--- a/atmel-samd/mpconfigport.h
+++ b/atmel-samd/mpconfigport.h
@@ -190,19 +190,16 @@ extern const struct _mp_obj_module_t usb_hid_module;
 #endif
 
 #define MICROPY_PORT_BUILTIN_MODULES \
-    { MP_OBJ_NEW_QSTR(MP_QSTR_touchio), (mp_obj_t)&touchio_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_microcontroller), (mp_obj_t)&microcontroller_module }, \
-    { MP_OBJ_NEW_QSTR(MP_QSTR_analogio), (mp_obj_t)&analogio_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_digitalio), (mp_obj_t)&digitalio_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_busio), (mp_obj_t)&busio_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_board), (mp_obj_t)&board_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_os), (mp_obj_t)&os_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_random), (mp_obj_t)&random_module }, \
-    { MP_OBJ_NEW_QSTR(MP_QSTR_storage), (mp_obj_t)&storage_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_time), (mp_obj_t)&time_module }, \
-    { MP_OBJ_NEW_QSTR(MP_QSTR_neopixel_write),(mp_obj_t)&neopixel_write_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_samd),(mp_obj_t)&samd_module }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_usb_hid),(mp_obj_t)&usb_hid_module }, \
+    { MP_OBJ_NEW_QSTR(MP_QSTR_audioio), (mp_obj_t)&audioio_module }, \
     EXTRA_BUILTIN_MODULES
 
 #define MICROPY_PORT_BUILTIN_DEBUG_MODULES \
```
And then I ran the example code from `audioio` documentation:

```
import audioio
import board
import array
import time
import math

# Generate one period of sine wav.
length = 8000 // 440
sine_wave = array.array("H", [0] * length)
for i in range(length):
    sine_wave[i] = int(math.sin(math.pi * 2 * i / 18) * (2 ** 15) + 2 ** 15)

sample = audioio.AudioOut(board.D1, sine_wave)
sample.play(loop=True)
time.sleep(1)
sample.stop()
```

This works — I could hear a sound on the speaker connected to D1. However, as soon as I do a soft reset of the board, the USB drive unmounts and the serial console closes, and when I re-connect to it, I get:

```
You are running in safe mode which means something really bad happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue here with the contents of your CIRCUITPY drive:
https://github.com/adafruit/circuitpython/issues
```
I suspect that there is some de-initialization thing that is not quite compatible with the Trinket M0, but looking at the code, I didn't find anything suspicious. I wonder how I could go about debugging this.

jerryneedell, 2017-09-24:
Did you try explicitly using deinit() at the end of your code or possibly use the context manager as noted in:  http://circuitpython.readthedocs.io/en/2.x/docs/design_guide.html#lifetime-and-contextmanagers. This was recommended in: http://circuitpython.readthedocs.io/en/2.x/shared-bindings/audioio/__init__.html

deshipu, 2017-09-24:
I just tried wrapping the code in a context manager, and that seems to be preventing the crash. I will try with the deinit now.

dhalbert, 2017-10-03:
Fixed by #304.

closed 2017-10-03 by dhalbert