adafruit/circuitpython #283: Rotary encoder support

tannewt, 2017-09-25:
SAMD51 has a peripheral for it.

sommersoft, 2018-02-18:
Just wanted to make some notes here. I was skimming through Noridc documentation, and noticed that nRF52x also has rotary encoder as a peripheral. Not surprising, since it's M4 based. 

They refer to it as "Quadrature decoder" or "QDEC". Here are links to the documentation:
- nRF52832: [QDEC](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52832.ps.v1.1/qdec.html?cp=2_1_0_35#concept_isq_pwd_4r)

- nRF52840: [QDEC](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52840.ps/qdec.html?cp=2_0_0_34#concept_isq_pwd_4r)

microbuilder, 2018-02-19:
The Arduino nRF52 Repo contains a [RotaryEncoder](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/RotaryEncoder) library with both HW and SW implementations for this, just as a reference. If a common API is added at the HAL layer, it shouldn't be too complicated to add nRF52 support used the QDEC peripheral or GPIO.

ladyada, 2018-02-19:
making a note here that SAMD has very fixed pins for QDEC - so for that platform, at least, we should just DIY with a timer interrupt - an existing one would probably be fine as the checking code is very small

tannewt, 2018-04-24:
Anyone mind if I punt this to 4.0?

commit d0fb6e7a (2018-06-01): atmel-samd: Add rotary encoder support.

Fixes #283
closed 2018-06-01 by dhalbert