adafruit/circuitpython #284: Ultimate GPS driver needed

hukuzatuna, 2017-09-27:
There is no current support for the Ultimate GPS boards (breakouts or Featherwings) in CircuitPython. We need to develop a buffered I/O library and command interface for sending and receiving NMEA data.

ladyada, 2017-09-30:
hiya do you mean parsing out the data? because we do have buffered UART I/O that works quite well:
https://learn.adafruit.com/adafruit-gemma-m0/circuitpython-uart-serial

hukuzatuna, 2017-09-30:
Yes, @ladyada, it's about creating the classes and methods to abstract the parsing from the user (e.g., breaking out the int into hours, minutes, seconds). This will allow users to make a call like "gps.getLat()" and receive a float decimal degree latitude in response. I currently have this built on top of the busio.UART interface.

ladyada, 2017-09-30:
ok whew - just makign sure!

caternuson, 2017-10-15:
> This will allow users to make a call like "gps.getLat()" and receive a float decimal degree latitude in response

Consider using `@property` instead of function call to be more pythonic. Returning both lat and lon in a tuple might be a nice way. Ex:
```python
lat, lon = gps.position
```

ladyada, 2017-11-28:
hiya this was done by @tdicola 
https://learn.adafruit.com/adafruit-ultimate-gps/circuitpython-parsing
huzzah!

closed 2017-11-28 by ladyada