adafruit/circuitpython #286: Make `storage.remount` accept `readonly` as a kwarg.

tannewt, 2017-09-27:
Right now you can only do `storage.remount("/", False)` when `storage.remount("/", readonly=False)` should be preferred.

To do this, we need to make the bindings for [`remount`](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/storage/__init__.c#L104) more like [`mount`](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/storage/__init__.c#L51).

However, the readonly kw in `remount` should not be marked `MP_ARG_KW_ONLY` in order to be backwards compatible.

process1183, 2017-10-08:
Hello,

I would like to work on this. I'll need some time to get up to speed on CircuitPython though. I should also have a [Feather M0 Express](https://www.adafruit.com/product/3403) to test on soon.

tannewt, 2017-10-09:
Great! That would be awesome @process1183. We chat regularly on our discord at http://adafru.it/discord where you can get help. We also have a weekly dev meeting you can join on discord Mondays at 11am PT/ 2pm ET.

tannewt, 2017-10-20:
@process1183 fixed this! Thank you!

closed 2017-10-20 by tannewt