adafruit/circuitpython #290: Support for APDS9960

dglaude, 2017-09-28:
I have an "APDS9960 Proximity, Light, RGB, and Gesture Sensor" and plan to do a minimal support library for Circuit Python (maybe just gesture detection).

This is a place holder to avoid duplicate work if somebody is working on this.
I will update if I make any progress... and if you start working on this, please say it here.

https://www.adafruit.com/product/3595

dglaude, 2017-09-28:
For reference, *Python ressources that can be usefull

Some very old attempt in Micro Python I made a year ago: https://gist.github.com/dglaude/1ab912dd060f22e54aa9dc083b058957

Some GPL code for plain Python found on Github for a simple version of the chip (9930 vs 9960):
https://github.com/Depaulicious/python-apds9930
+
https://gist.github.com/Depaulicious/08689d6d1e58f7700e4a

Some Python project that failed to use apds9960 because of missing library:
https://www.hackster.io/bastiaan-slee/nabaztag-gets-a-new-life-with-google-aiy-e9f2c8

dglaude, 2017-09-28:
I created a repository for this library: https://github.com/dglaude/adps9960-circuit-python

tannewt, 2017-10-02:
Cool! Ping me when you want me to look at it. Once its ready I'll fork it into adafruit's repos.

ladyada, 2017-11-28:
oof we have to issues for this, other one is here!
https://github.com/adafruit/circuitpython/issues/411

dglaude, 2017-11-28:
I have not started working on the APDS9960 driver for circuit python...
So if anybody feel like doing it, please do.
I'll report here if I find more time and energy for that project.
Regards

dglaude, 2017-12-17:
Some work in progress from @mrmcwethy here: https://github.com/mrmcwethy/Adafruit_CircuitPython_APDS9960

mrmcwethy, 2017-12-31:
I have completed a candidate library that is written CircuitPython that support gesture, proximity, and color detection.  I used the Adafruit_APDS9960 Arduino library as a template for the CP version.  This has had some serious testing by @jerryn and @kattni .  Thanks to them.  i would like the following project to be reviewed and forked into github.com/adafruit.   @tannewt  has already given me early feedback and I have made the requested adjustments as a result.

https://github.com/mrmcwethy/Adafruit_CircuitPython_APDS9960

ladyada, 2018-01-02:
@deanm1278 heya dean, here's the py library to test :)

tannewt, 2018-01-02:
I've forked it into the Adafruit repo. Please do a release and add it to
the bundle. Thanks!

deanm1278, 2018-01-02:
@ladyada @mrmcwethy @tannewt Tested all examples with circuitpython metro m0 2.1.0, latest bundle 2.1.0, myp-cross 2.1.0
gesture, color, prox interrupt all work very well! Interface is very good any circuitpythony

dglaude, 2018-01-02:
Dear @mrmcwethy , I tested a little bit on
`Adafruit CircuitPython 2.2.0-rc.2 on 2017-12-30; Adafruit Feather M0 Express with samd21g18`

Testing involved:
1) Copy the 3 files `gesture.py` `proximity.py` and `color.py` from https://github.com/mrmcwethy/Adafruit_CircuitPython_APDS9960/tree/master/adafruit_apds9960 into a folder named `/Volumes/CIRCUITPY/lib/adafruit_apds9960`
2) Copy the 3 examples files 
`/Volumes/CIRCUITPY/`
3) Copy each to main.py to test them one by one.

They all fail with `MemoryError: memory allocation failed, allocating ### bytes` at respectively line 3, 4 and 5.
The offending line is `from adafruit_apds9960.apds9960 import APDS9960`.

Since @deanm1278 successfully tested, I may I missed one step.
Should I use myp-cross to consume less memory?

dglaude, 2018-01-02:
My fault... I had to use mpy-cross-2.2.0-macos-high-sierra to transform __init__.py apds9960.py colorutility.py into *.mpy file.

Now I can confirm it is working with 2.2.0-rc.2.

The only issue I have is on the non uniform use of pin for interrupt.

`Readme.rst` suggest to use A1:
```
int_pin = digitalio.DigitalInOut(A1)
```

`Proximity.py` example use A2:
```
int_pin = digitalio.DigitalInOut(board.A2)
```

Some cleaning to make Readme and example match in style and pin selection would be welcome.

mrmcwethy, 2018-01-02:
I agree.  i fixed  it and created a pull request.  Next is to create a release and add it to the bundle.

ladyada, 2018-01-02:
ill do that part!

tannewt, 2018-01-05:
This is done.

closed 2018-01-05 by tannewt