adafruit/circuitpython #291: Merge in nrf52 support.

tannewt, 2017-09-28:
From pull https://github.com/micropython/micropython/pull/3137 if its not pulled upstream.

ladyada, 2017-10-23:
1) serial repl!
2) digitalio module
3) time module
4) local flash storage - 128KB of internal flash

microbuilder, 2017-11-01:
Local flash storage will have to be reduced to ~92KB assuming the following:

- 12KB Master Boot Record
- 48KB Serial Bootloader
- 140KB SoftDevice S132v5
- 28KB NFFS-based internal config memory (bonding data, etc.)
- 192KB CircuitPython Code

Remaining Space: 512KB - 420KB = **92KB local flash storage**

ladyada, 2017-11-01:
sounds fine to me - thats enough to get people going. our final design will have external SPI flash :)

tannewt, 2017-11-01:
A couple naive questions from me:

* Why is the MBR 12k? [Here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.1%2Fbledfu_memory.html) its only 4k.
* Why is the bootloader 48k? Bossac and UF2 are only ~8k.
* Can we combine the NFFS filesystem with the file system for CircuitPython? Without USB we can use NFFS but we'll want FATFS for the USB version.

microbuilder, 2017-11-02:
- For the MBR in that link 0x0..0x3000 = 12KB (???)
[image]
- The bootloader also supports OTA updates via BLE, which is why it's so large
- I will probably combine the two file systems internally yes (nRF52832 variants without USB), and simply hide the config files, but I always need to be sure that we have 'n' bytes free for storing the bonding data for up to 5 devices and other BLE config values, which is why for the moment I've kept them separate. This can be reduced to something smaller than 28KB, I just don't have the exact lowest safe number yet.

microbuilder, 2017-11-02:
There is conflicting information about the sizes in the documentation. :(

I've listed below what seems to be the right/final settings for:

- S132 v2.0.1 (current feather52 boards)
- S132 v5.0.0 (future work and nRF52840)

## S132 v2.0.1 with OTA DFU Bootloader

- CircuitPython Core: **228KB**
- User Filesystem: **96KB**
- Private Config Data: **28KB** (which may get combined for 96+28=**124KB** total)

```
/*
    GNU linker script for NRF52 w/ s132 2.0.1 SoftDevice

    MEMORY MAP
    ------------------------------------------------------------------------
    START ADDR  END ADDR   SIZE    DESCRIPTION
    ----------  ---------- ------- -----------------------------------------
    0x00074000..0x00080000 ( 48KB) Serial + OTA Bootloader
    0x0006D000..0x00073FFF ( 28KB) Private Config Data (Bonding, Keys, etc.)
    0x00055000..0x0006CFFF ( 96KB) User Filesystem
    0x0001C000..0x00054FFF (228KB) Application Code
    0x00001000..0x0001BFFF (108KB) SoftDevice
    0x00000000..0x00000FFF   (4KB) Master Boot Record
*/
```

## S132 v5.0.0 with OTA DFU Bootloader

The 5.0.0 SoftDevice adds Bluetooth 5.0 support, and enables long packets (>20 bytes), but requires more flash memory to use.

- CircuitPython Core: **200KB**
- User Filesystem: **96KB**
- Private Config Data: **28KB** (which may get combined for 96+28=**124KB** total)

```
/*
    GNU linker script for NRF52 w/ s132 5.0.0 SoftDevice

    MEMORY MAP
    ------------------------------------------------------------------------
    START ADDR  END ADDR   SIZE    DESCRIPTION
    ----------  ---------- ------- -----------------------------------------
    0x00074000..0x00080000 ( 48KB) Serial + OTA Bootloader
    0x0006D000..0x00073FFF ( 28KB) Private Config Data (Bonding, Keys, etc.)
    0x00055000..0x0006CFFF ( 96KB) User Filesystem
    0x00023000..0x00054FFF (200KB) Application Code
    0x00001000..0x00022FFF (136KB) SoftDevice
    0x00000000..0x00000FFF   (4KB) Master Boot Record
*/
```

tannewt, 2018-01-16:
This is done. Its in master and has other issues for its additional polish.

closed 2018-01-16 by tannewt