adafruit/circuitpython #297: touchio (capacitive touch) on Gemma #A0 or Trinket #D1 not always firing

ladyada, 2017-09-30:
i can't seem to consistantly get touch.value to ever be True on a trinket or gemma (running 2.0.0) this has happened to me before too....

could it be a threshhold thing? is there any way to expose the raw_value? maybe could help debug

dhalbert, 2017-09-30:
I just tested on a Gemma and Trinket with the same pins, and it's working for me on a wooden desk with my finger directly on the pad or with an alligator lead. Code is just something like:
```
import board
import touchio
t = touchio.TouchIn(board.D1)
t.value
```
I put it in a loop, etc., and it was OK. Anything else going on?

We have an old pull request from a user for the raw value that was never finished (#136). We could revive that easily or do something similar.

ladyada, 2017-09-30:
i grabbed another trinket and on that one, D1 works but not D3/4 - but those pins definitely work for other stuff. so, i kinda suspect a threshhold thing?
i'd dig having the raw_value if we can fit it in

dhalbert, 2017-09-30:
I sent @ladyada some test builds with .raw_value available.

ladyada, 2017-09-30:
yeah the 'default raw' value is 2000 for D1 and ~1500 for the others. you can see how D1 doubles when i touch it, but maybe because it is higher than D3/D4 it doesnt trigger?

```
# Trinket IO demo - captouch to dotstar

import touchio
import busio
import board
import time

touch0 = touchio.TouchIn(board.D1)
touch1 = touchio.TouchIn(board.D3)
touch2 = touchio.TouchIn(board.D4)

dotstar = busio.SPI(board.APA102_SCK, board.APA102_MOSI)

r = g = b = 0
 
def setPixel(red, green, blue):
    if not dotstar.try_lock():
        return
    print("setting pixel to: %d %d %d" % (red, green, blue))
 
    data = bytearray([0x00, 0x00, 0x00, 0x00,
                      0xff, blue, green, red,
                      0xff, 0xff, 0xff, 0xff])
    dotstar.write(data)
    dotstar.unlock()
    time.sleep(0.01)
 
while True:
    print("%d %d %d" % (touch0.raw_value, touch1.raw_value, touch2.raw_value))
    if touch0.value:
        r = (r+1) % 256
    if touch1.value:
        g = (g+1) % 256
    if touch2.value:
        b = (b+1) % 256
 
    setPixel(r, g, b)
    time.sleep(0.1)
```

> Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
> main.py output:
> 2039 1288 1347
> setting pixel to: 0 0 0
> 2041 1285 1349
> setting pixel to: 0 0 0
> 2040 1292 1346
> setting pixel to: 0 0 0
> 2038 1285 1348
> setting pixel to: 0 0 0
> 2040 1291 1347
> setting pixel to: 0 0 0
> 2038 1285 1347
> setting pixel to: 0 0 0
> 2040 1285 1346
> setting pixel to: 0 0 0
> 2039 1284 1346
> setting pixel to: 0 0 0
> 2043 1286 1347
> setting pixel to: 0 0 0
> 2040 1286 1346
> setting pixel to: 0 0 0
> 2036 1286 1348
> setting pixel to: 0 0 0
> 2038 1291 1350
> setting pixel to: 0 0 0
> 2039 1288 1345
> setting pixel to: 0 0 0
> 2038 1289 1346
> setting pixel to: 0 0 0
> 2038 1289 1346
> setting pixel to: 0 0 0
> 3955 1289 1348
> setting pixel to: 0 0 0
> 3996 1290 1352
> setting pixel to: 0 0 0
> 4043 1290 1350
> setting pixel to: 0 0 0
> 2038 1290 1346
> setting pixel to: 0 0 0
> 3774 1286 1348
> setting pixel to: 0 0 0
> 3990 1294 1350
> setting pixel to: 0 0 0
> 3982 1292 1349
> setting pixel to: 0 0 0
> 4043 1290 1349
> setting pixel to: 0 0 0
> 2037 1285 1349
> setting pixel to: 0 0 0
> 2038 1289 1346
> setting pixel to: 0 0 0
> 2042 1289 1346
> setting pixel to: 0 0 0
> 2039 1285 1346
> setting pixel to: 0 0 0
> 2037 1285 1346
> setting pixel to: 0 0 0
> 2042 1287 1348
> setting pixel to: 0 0 0
> 2042 1285 1349
> setting pixel to: 0 0 0

dhalbert, 2017-09-30:
The threshold for `.value == True` is twice the initial reading when the object is constructed.

ladyada, 2017-09-30:
that is waay lower than the threshhold ASF uses - they use maybe like 64 points

dhalbert, 2017-10-01:
>that is waay lower than the threshhold ASF uses - they use maybe like 64 points

Not sure I know what you mean here. Do you mean the raw value range is only 0-63?

 The current algorithm is to read the raw value on construction, and then double it for the True/False threshold. E.g., if the initial raw value at construction is 2042, the threshold is 4084. Your worst case was 3774/2038 = 1.85. We could lower the threshold to 1.75*the initial raw value.

On my samples, the ratio is always well above 2, but that could have to do with the humidity, etc.

We could expose one or more properties, like allowing the user to get/set the threshold and/or the ratio. That does expose the algorithm, which we might want to change. @tannewt, do you have anything to add here? How did you choose 2?

ladyada, 2017-10-01:
oops i mean, like 64 points *above* baseline

dhalbert, 2017-10-01:
I tried a Makey-Makey style use:

Banana, clip lead on stem, no-touch / touch:
D1: 3573 / 4064
D3: 2763 / 4064
D4: 2879 / 4064

Apple, clip lead on stem
D1: 3000 / 3891
D3: 2183 / 3694
D4: 2257 / 3727

Clementine well past its prime:
D1: 2896 / 4064
D3: 2231 / 4064
D4: 2187 /4064

I think this suggests we should use an absolute increase above baseline rather than a ratio. Also maybe we really should expose the threshold for getting/setting.

I don't know why D1 has a consistently higher initial value than D3 and D4. I double-checked in the REPL to make sure no pull-up/down was set initially, etc.

ladyada, 2017-10-01:
D1 is shared with the DAC circuitry so i think it has a different built in capacitance

tannewt, 2017-10-02:
Absolute increase works for me! Thanks for the investigation @ladyada and
@dhalbert.

PR #300 (2017-10-02): Make touch more sensitive. Add .raw_value and .threshold attributes.
Fixes #297. Makes the threshold be 100 above the initial value, instead of 2 times the initial value. Works better for attached objects such as fruit. Add `.raw_value` and `.threshold` attributes to allow debugging and fine-tuning of touch sensitivity.
closed 2017-10-03 by dhalbert