adafruit/circuitpython #298: ridiculous idea: 'breakpoints'

ladyada, 2017-09-30:
this doesnt exist in python i think but its an interesting idea - can we pop out from the main.py code to REPL but still have access to that context?

tannewt, 2017-09-30:
We could. Micropython actually worked that way originally where the heap
was shared. There was no way to continue main though. Would the goal be for
debugging?

ladyada, 2017-09-30:
yeah like "stop here i want to see my variable state" - im ok with not being able to re-continue

caternuson, 2017-10-15:
> this doesnt exist in python i think
try `python -i foo.py`

dhalbert, 2017-10-25:
A PEP (Python Enhancement Proposal) for a builtin function called `breakpoint()`: https://www.python.org/dev/peps/pep-0553/. Will probably be in Python 3.7.

dhalbert, 2017-10-25:
yeah - it provides us a syntax that will be the standard

tannewt, 2017-10-25:
We could also reach out upstream to see if its development we could fund.

dastels, 2017-11-02:
I've started looking into this. Initially I'm playing in the unix port, but the code should all live in py/

https://github.com/dastels/circuitpython

ladyada, 2017-11-02:
rad. to extend my terrible idea...would it ever make sense to have a debugging backchannel thru HID...so REPL stays clean

dastels, 2017-11-02:
I think then you start thinking about watchpoints, tracing, single stepping, and the like. And that will need far more invasive changes. Certainly doable, but might not be in line with a simple, educational environment.  

My goal is to be able to drop in a debugger() or breakpoint() (Smalltalk used halt() ) function call and let you poke around the local environment, possibly peek into different frames on the stack.  I'd also like to support passing an boolean expression to debugger() (or whatever) to allow conditional breakpoints. That should be trivial to add once the basics are in place.

ladyada, 2017-11-02:
yah agree- i was just thinking to the far future :) for now breakpoints are enough!

dastels, 2017-11-02:
I'll probably explicitly NOT support using `def` in the debug repl. Expressions, yes. Assignment to in-scope variables, yes with a warning that it might mess things up. I'd like to support looking at variables in prior stack frames. I'm not sure how that would look yet... but that's not for a while.  The next step is just to get local variable (and parameter) references working.

tdicola, 2017-11-02:
Would re-enabling stopping of main and keeping the context be a simple interim fix in 3.0 or earler?  It's handy to debug your main.py by Ctrl-C and looking at variable values.

dastels, 2017-11-03:
My first thought it that where you stop it would be random. Even in a simple loop where you stop would impact what you expect values to be.

ScottDWebster, 2018-05-21:
It would be nice if there was something like DBGP (https://xdebug.org/docs-dbgp.php) to allow remote debugging.

hybotics, 2019-08-14:
Hi,

I am VERY interested in seeing some kind of source level debugging added to Circuitpython. By this I mean the facility to have a script stop at a certain point (breakpoint), be able to examine and change variables, and the continue after the breakpoint. I am at a point with one script where it would be VERY useful to be able to do this.

For instance, I have verified that code sequences work, by using the REPL, but I am not able to see what my script is actually doing and why it is not working right. This is where I think source level debugging would come in VERY handy. Print statements just are not giving me what I need.

8-Dale (aka geekguy on the forums)

kevinjwalters, 2020-03-13:
I did a search on forums and found [Adafruit Forums: Is using circuitpython on rPi the only way to use a debugger](https://forums.adafruit.com/viewtopic.php?f=60&t=155219).

I have some moderately complex code now in CircuitPython and this has some bugs in it. Normally this would be when I would add a breakpoint and then single-step the (python) code and I think I'd be able to work it out in a few minutes...

I think this is going to become more common and more of a gap as the bigger CPUs like SAMD51 and nRF52840 allow more complex programs to be written.

canyon289, 2020-11-05:
Support for an interactive repl debugger like `breakpoint()` or `import pdb; pdb.set_trace()' would be nice, but I can get away with print statements in my case for now :)

calcut, 2022-02-02:
Just adding my interest in this feature.

ladyada, 2022-07-11:
https://www.youtube.com/watch?v=RRN_ngx01vM