adafruit/circuitpython #299: CPX A0 is not reliable for touchio.TouchIn

dhalbert, 2017-10-01:
CPX A0 is connected to the speaker, so it will not work for touch input if the speaker is in use. This should be documented if it's not already

In addition, `Touchio.TouchIn(board.A0)` often ends up with a very high initial `threshold` value when the object is created. If you `deinit()` the object and re-create, the `threshold` is more reasonable and is closer to the `raw_value` read later. I tried reading the value twice on during construction but it was still high to being with.

caternuson, 2017-10-15:
The new guide does a good job of documenting this:
https://learn.adafruit.com/adafruit-circuit-playground-express/pinouts

dhalbert, 2017-10-15:
Should we just remove A0 from the allowed pins for TouchIn?

kattni, 2017-10-15:
That would be my suggestion. I specifically didn't include it in the touch API because of the inconsistency.

dhalbert, 2017-10-15:
Turns out the current implementation does not actually check the chosen pin against a list of allowed pins.

kattni, 2017-10-15:
That would be my suggestion. I specifically didn't include it in the CPX touch API because of the inconsistency.

dhalbert, 2017-10-16:
Fixed by #330. A0 is now not allowed for TouchIn, only on CPX.

closed 2017-10-16 by dhalbert