adafruit/circuitpython #307: unable to terminate sys.stdin.readlines() from terminal session

jerryneedell, 2017-10-04:
entering the following:
```
import sys
sys.stdin.readlines()

```
then typing some input and a Control-D  does not terminate the input. It just hangs .
Control-C terminates it, but the input is lost.

also tried as main.py with same result.

The same sequence works as expected in a pythpn3 session on my desktop.

dhalbert, 2017-10-04:
`mp_hal_stdin_rx_chr()` just passes ctrl-D through. `input()` handles it specially and throws `EOFError` if it encounters a ctrl-D at the beginning of a line. `readlines()` does not do anything with it, which makes sense, since there might be a ctrl-D in the middle of a line of a file.

dhalbert, 2017-10-08:
The way this works is inherited from MicroPython. I opened an [issue](https://github.com/micropython/micropython/issues/3356) in their repo to see if they are aware of this and what their position would be.

ladyada, 2021-02-20:
do we want to close this as 'wontfix' since we have dual CDC support now and honestly thats a way way better system for reading lines?

dhalbert, 2021-02-20:
The original thing of using `readlines()` is still a problem, but the secondary CDC can substitute, and you can signal the end of the read however you want. Plus there are timeouts, and you can ask how many characters are left to read, etc. So, yes, I'm closing, but @jerryneedell, if you still want some other semantics, feel free to reopen.

closed 2021-02-20 by dhalbert