adafruit/circuitpython #308: ValueError causing Circuit Playground Express to crash to safe mode

kattni, 2017-10-04:
Making specific changes to the express class (attached [express.txt](https://github.com/adafruit/circuitpython/files/1357363/express.txt)) in the Circuit Playground Express library causes the CPX to reboot into safe mode. 

As is, the `code.py` (attached [code.txt](https://github.com/adafruit/circuitpython/files/1357360/code.txt)) picks up the current reading of x, y, z orientation and lights up the LEDs the corresponding color, and prints and repeats that reading indefinitely in the serial REPL.

To cause the crash:
On line 84, delete `self.x, ` and uncomment lines 86 and 87. Save. Once out of 3 times, I had to save `express.py` twice before it failed. Here is the last bit from the REPL, every time it fails

```-4.13791,  0.0,  -8.7356
-4.13791,  0.0,  -8.7356
-4.13791,  0.0,  -8.7356
Traceback (most recent call last):
  File "code.py", line 49, in <module>
KeyboardInterrupt:
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "/lib/adafruit_circuitplayground/express.py", line 294, in <module>
  File "/lib/adafruit_circuitplayground/express.py", line 84, in __init__
ValueError: too many values to unpack (expected 2)

Press any key to enter the REPL. Use CTRL-D to reload.
```
It then disconnects completely from the REPL back to the computer's command line prompt.

dhalbert, 2017-10-05:
I was able to reproduce this. But I think it's due to delayed writes, and not some particular issue with this code. I was in a  "safe mode" state which recurs again and again if I type ctrl-D. But once I did a hard reset, everything was fine.

closed 2017-10-07 by dhalbert

violentlyhappi, 2017-12-02:
I'm a newbie with this board. How do you hard reset it? I keep getting a file storage issue and I only have the main.py, lib folder (which is empty), and boot_out.txt. Thanks!

dhalbert, 2017-12-02:
@violentlyhappi If you have drive issues, see this link for how to reset the CIRCUITPY filesystem: https://learn.adafruit.com/adafruit-circuit-playground-express/troubleshooting#circuitpy-drive-issues

These GitHub issues are more for bugs and feature requests. For support, post in the support forums: https://forums.adafruit.com, or see the #circuitpython channel on our chat server: https://adafru.it/discord.