adafruit/circuitpython #310: transfer support for SPI devices

deanm1278, 2017-10-09:
We need a way to do a full duplex SPI transfer. It would read and write at the same time

would likely go something like:

`
c = None
with self.spi_device as spi:
			c = spi.transfer(bytearray([0x0A, 0x04]))`

c would be a bytearray of the data read during the transfer

tannewt, 2017-10-09:
How many bytes at a time do you want to do? Are you going to turn around
and send the previously received byte? Doing that through python will be
very slow.
~Scott

deanm1278, 2017-10-09:
for this specific application yea I'm doing one byte at a time and sending the previously received byte so it would be something like:

```
with self.spi_device as spi:
   c = bytearray( [0x0A] )
   for i in range(n):
      c = spi.transfer( c  )
```
it will be slow, but this e-paper display takes around 10 sec to update anyway so it's acceptable. In general though we probably want to be able to transfer as many bytes as is allowed by available RAM.

tannewt, 2017-10-10:
If you aren't worried about speed is it required to simultaneously transfer
in both directions? Can you just do a read and then a write?

deanm1278, 2017-10-10:
yea unfortunately it needs to transfer simultaneously in both directions in this application. 
although full-duplex communication isn't used all that often it is kinda just part of SPI spec and will need to be implemented eventually :/

tannewt, 2017-10-10:
Ok good to know!

deanm1278, 2017-10-10:
for this application though, the reason it needs to be done full-duplex is because the EPD driver expects all data to be written in one transaction. Since we don't have enough on-chip RAM available to store the entire buffer, we need to store it on an external SRAM chip and "read to" the EPD from there.
Both the EPD driver and the SRAM slave devices are active on the bus at one time

tannewt, 2017-10-10:
We don't need transfer to use DMA for now. We can add that in 3.x.

dhalbert, 2017-10-10:
@deanm1278 I can probably take this on, but I want to order something that needs this, is  https://www.adafruit.com/product/3502 (Sharp e-ink breakout) suitable? I'll ask you for your driver code when I'm ready to test.

tannewt, 2017-10-10:
You shouldn't actually need any hardware. It should be enough to tie MISO high or low and then use the saleae to verify MOSI output.

deanm1278, 2017-10-10:
thanks @dhalbert!

you can do what @tannewt says or you can use this code on an Atmega 328 based board (metro, UNO)

```
void setup (void)
{
  pinMode(MISO, OUTPUT);

  // turn on SPI in slave mode
  SPCR |= bit(SPE);

  // turn on interrupts
  SPCR |= bit(SPIE);
}

// SPI isr
ISR (SPI_STC_vect)
{
  byte c = SPDR;
  
  SPDR = c;
} 

void loop (void)
{
  //do nothing
}
```

wire up the cirpy -> metro as:
SCK -> 13
MISO -> 12
MOSI -> 11
CS -> 10

the metro will act as an SPI slave and echo back the last byte that was sent, so you can verify both receive and transmit.

here is the SPI master code I used to test:

```
#include <SPI.h>

const int slaveSelectPin = 10;

uint8_t pos = 0;

void setup() {
  pinMode(slaveSelectPin, OUTPUT);
  SPI.begin();
}

void loop() {
  digitalWrite(slaveSelectPin, LOW);
  uint8_t i = SPI.transfer(pos);
  digitalWrite(slaveSelectPin, HIGH);
  Serial.println(i);
  pos++;
  delay(500);
}
```

EDIT: updated with a slightly better way to test

dhalbert, 2017-10-16:
I'm in progress on this. Do we want bidirectional support for `bitbangio` as well at the same time, or can that be deferred a bit? I don't think it's that hard, but the `shared-module` implementation could probably use some refactoring to help supporot it.

tannewt, 2017-10-16:
I think its ok to wait as long as native SPI on ESP8266 works. Make sure
the API is there though, just have it return NotImplementedError. Thanks!

dhalbert, 2017-12-06:
Implemented by #471 (samd) and #473 (esp8266).

closed 2017-12-06 by dhalbert