adafruit/circuitpython #312: Add driver for ADS1x15 ADCs

caternuson, 2017-10-09:
Add driver to support the ADS1x15 ADCs.
- ADS1015 12-Bit ADC ([PID 1083](https://www.adafruit.com/product/1083))
- ADS1115 16-Bit ADC ([PID 1085](https://www.adafruit.com/product/1085))

tannewt, 2017-10-10:
I'd love it if ADC drivers exposed a number of sub-objects for each input that can be used in place of an [`AnalogIn`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/analogio/AnalogIn.html). To do that, it'd need the same properties.

tannewt, 2017-10-11:
my goal would be able to do something like:

```python
# setup analog

# import analogio
# import board
# a = analogio.AnalogIn(board.A0)

# OR!
import adafruit_CircuitPython_ADS1x15
adc = adafruit_CircuitPython_ADS1x15.ADS1015()
a = adc[0]

while True:
    # change neopixel color based on analog readings from a
```

That way all code written for one analog input is agnostic to whether its on the MCU or on a helper chip.

caternuson, 2017-10-11:
In-work version is here:
https://github.com/caternuson/CircuitPython_ADS1x15

caternuson, 2017-10-11:
Something like this (but with a generator instead of static list of channels in the ADC class):

```python
import random

class ADC_Channel():
    def __init__(self, adc, channel):
        self._adc = adc
        self._channel = channel
        self._value = None
    
    @property
    def value(self, ):
        #self._value = random.random()
        self._value = self._adc.read_adc(self._channel)
        return self._value 

class Generic_ADC():
    def __init__(self, ):
        self.channels = [ADC_Channel(self, 0),
                         ADC_Channel(self, 1),
                         ADC_Channel(self, 2),
                         ADC_Channel(self, 3)]
    
    def __getitem__(self, key):
        return self.channels[key]
    
    def read_adc(self, chan):
        return chan + random.random() 

adc = Generic_ADC()
a = adc[3]
a.value
```

tannewt, 2017-10-11:
Yup! Thats what I was thinking.

caternuson, 2017-10-11:
Could get a little messy trying to support all the modes the ADC can run in: single ended, differential, comparator. AnalogIn is essentially single ended only, so could make it to where that is the only mode supported for the above access method.

The ADC class will still have full functionality.
```python
adc.read_adc()
adc.read_adc_difference()
adc.start_adc_comparator()
# etc
```

tannewt, 2017-10-11:
Sure. The other way to do it would be three separate drivers for the three
different modes. Each could be in a different file so you only need to load
the code related to your use.

caternuson, 2017-10-11:
Separate ADC classes?

tannewt, 2017-10-11:
Yup, thats what I mean. So maybe:

```
from adafruit_ADS1x15.single_ended import ADS1015
```

What do you think? The model of more, smaller classes makes sense in terms of reducing memory footprint.

caternuson, 2017-10-11:
Hmmm. Or if we add a ```mode``` property to the ADC class. Could then do something like:
```python
class ADC_Channel():
    def __init__(self, adc, channel):
        self._adc = adc
        self._channel = channel
        self._value = None
    
    @property
    def value(self, ):
        if self._adc._mode == self._adc.SINGLE_ENDED:
            self._value = self._adc.read_adc(self._channel)
        elif self._adc._mode == self._adc.DIFFERENTIAL:
            self.value = self._adc.read_adc_difference(self._channel)
        elif self._adc._mode == self._adc.COMPARATOR:
            self.value = self._adc.get_last_result()
        return self._value 
```

caternuson, 2017-10-11:
I like keeping ADC to just one class - or more importantly, making the CircuitPython version match the Python version as much as possible. Otherwise there would be an Arduino way of using it, a Python way of using it, and a CircuitPython way of using it.

I'm thinking this feature should be, as much as possible, a bolt on to the ADC class.

caternuson, 2017-10-11:
How about this? (mode is in channel, not adc)
```python
import random

class ADC_Channel():
    
    SINGLE_ENDED = 0
    DIFFERENTIAL = 1
    COMPARATOR   = 2
    
    def __init__(self, adc, channel):
        self._adc = adc
        self._channel = channel
        self._mode = ADC_Channel.SINGLE_ENDED
        self._value = None
        
    @property
    def mode(self, ):
        return self._mode
    
    @mode.setter
    def mode(self, mode):
        self._mode = mode
    
    @property
    def value(self, ):
        if self.mode == ADC_Channel.SINGLE_ENDED:
            self._value = self._adc.read_adc(self._channel)
        elif self.mode == ADC_Channel.DIFFERENTIAL:
            self._value = self._adc.read_adc_difference(self._channel)
        elif self.mode == ADC_Channel.COMPARATOR:
            self._value = self._adc.get_last_result()
        return self._value 

class Generic_ADC():
    def __init__(self, ):
        self.channels = [ADC_Channel(self, 0),
                         ADC_Channel(self, 1),
                         ADC_Channel(self, 2),
                         ADC_Channel(self, 3)]
    
    def __getitem__(self, key):
        return self.channels[key]
    
    def read_adc(self, chan):
        return chan + random.random()
    
    def read_adc_difference(self, chan):
        return -1 * (chan + random.random())
    
    def get_last_result(self, ):
        return random.random()

adc = Generic_ADC()
# use ADC directly (normal)
print adc.read_adc(1)
print adc.read_adc_difference(1)
print adc.get_last_result()
# get an ADC channel to use as an AnalogIn replacement
# this is probably the most common usage
a = adc[0]
print a.value
# support for differential mode
a = adc[1]
a.mode = ADC_Channel.DIFFERENTIAL
print a.value 
# support for comparator mode
a = adc[2]
a.mode = ADC_Channel.COMPARATOR
print a.value
```

tannewt, 2017-10-11:
I think keeping the circuitpython version like the python version is misleading. Our hope is to go back and implement the lower level APIs so circuitpython libraries work on raspberry pi as well.

Arduino and Python are different enough that its worth redesigning the API to fit python better.

tannewt, 2017-10-11:
I don't like mode as a channel property because modes affect the number of channels. Right?

It also changes the meaning of `value` which adds a ton of complexity to the API because you can't understand value without reading mode as well.

tannewt, 2017-10-11:
Ah ok. I still don't like the meaning of `value` depending on `mode`.

caternuson, 2017-10-12:
Does AnalogIn have a differential capability? Like, what is the differential form of this:
```python
import analogio
import board
a = analogio.AnalogIn(board.A0)
```

dhalbert, 2017-10-12:
Nope, no differential support, though there could be. If you have a proposed API that would be great (open a new issue). I briefly looked in the Arduino world and I didn't really see anything.

tannewt, 2017-10-12:
I'd probably do it as a separate class because these APIs are functionally split rather than split by peripheral.

tannewt, 2017-10-23:
Where are we at on this @caternuson ? It doesn't look like [this code](https://github.com/caternuson/CircuitPython_ADS1x15/blob/master/adafruit_ads1x15.py) has any notion of channels.

tannewt, 2017-10-24:
FYI, newer code is here: https://github.com/caternuson/CircuitPython_ADS1x15/tree/multi

caternuson, 2017-10-25:
Yeah, made a new branch to work on the approach of breaking into multiple files. Sorry if that was confusing. "Master" is now like "The original idea" or something.

caternuson, 2017-11-07:
PR made:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/pull/1

closed 2017-11-09 by caternuson