adafruit/circuitpython #313: atmel-samd: Non-fatal warnings when building master/3.0

tdicola, 2017-10-10:
I noticed some non-fatal warnings when building the current master/3.0 branch for circuit playground express:
```
arm-none-eabi-gcc -Os -DNDEBUG -finline-limit=49 -DNDEBUG -flto -finline-limit=49 -I. -I.. -I../lib/mp-readline -I../lib/timeutils -Iasf4/samd21 -Iasf4/samd21/hal/include -Iasf4/samd21/hal/utils/include -Iasf4/samd21/hri -Iasf4/samd21/hpl/core -Iasf4/samd21/hpl/pm -Iasf4/samd21/hpl/port -Iasf4/samd21/hpl/tc -Iasf4/samd21/include -Iasf4/samd21/CMSIS/Include -Iasf4/samd21/usb -Iasf4/samd21/usb/class/cdc -Iasf4/samd21/usb/class/hid -Iasf4/samd21/usb/device -Iasf4_conf/samd21 -Iboards/circuitplayground_express -Iboards/ -Ifreetouch -Ibuild-circuitplayground_express -Wall -Werror -std=gnu11 -nostdlib -fsingle-precision-constant -fno-strict-aliasing -Wdouble-promotion -Wno-endif-labels -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -Wfloat-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wnested-externs -Wunreachable-code -Wcast-align -Wno-error=lto-type-mismatch -D__SAMD21G18A__ -DCONF_USB_COMPOSITE_IDPRODUCT=0x8019 -DCONF_USB_COMPOSITE_IDVENDER=0x239A -ffunction-sections -fdata-sections -fshort-enums --param max-inline-insns-single=500 -DFFCONF_H=\"lib/oofatfs/ffconf.h\"  -mthumb -mabi=aapcs-linux -mcpu=cortex-m0plus -msoft-float -mfloat-abi=soft -DSAMD21 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -Wno-error=lto-type-mismatch   -c -o build-circuitplayground_express/frozen_mpy.o build-circuitplayground_express/frozen_mpy.c
usb.c:108:18: warning: type of '_usb_ep1_cache' does not match original declaration [-Wlto-type-mismatch]
 extern uint32_t *_usb_ep1_cache;
                  ^
asf4/samd21/hpl/usb/hpl_usb.c:303:10: note: '_usb_ep1_cache' was previously declared here
 uint32_t _usb_ep1_cache[_usb_align_up(CONF_USB_EP1_CACHE) / 4];
          ^
asf4/samd21/hpl/usb/hpl_usb.c:303:10: note: code may be misoptimized unless -fno-strict-aliasing is used
build-circuitplayground_express/frozen_mpy.c:54:26: warning: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Wlto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../py/qstr.c:98:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../py/qstr.c:113:26: warning: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Wlto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-circuitplayground_express/frozen_mpy.c:55:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^

9056 bytes free in flash out of 187904 bytes ( 183.5 kb ).
7240 bytes free in ram for stack out of 32768 bytes ( 32.0 kb ).

Create build-circuitplayground_express/firmware.bin
Create build-circuitplayground_express/firmware.uf2
python2 ../tools/uf2/utils/uf2conv.py -c -o build-circuitplayground_express/firmware.uf2 build-circuitplayground_express/firmware.bin
Converting to uf2, output size: 357888, start address: 0x2000
Wrote 357888 bytes to build-circuitplayground_express/firmware.uf2.
```

Opening to keep track and push for zero warnings with 3.0.

dhalbert, 2017-10-11:
I made the `_usb_ep1_cache` cache warning go away by declaring it in `usb.c` as:
`extern uint32_t _usb_ep1_cache[];`
instead of 
`extern uint32_t *_usb_ep1_cache;`

The current declaration is a part of a hack due to some bug or other issue with the ASF4 USB stack. @tannewt is reaching into an internal data structure to get the input characters.

The other warning is explained in #248: it's a gcc LTO bug. The LTO folks are testing a fix, so we should see it eventually.

dhalbert, 2018-01-09:
First warning is fixed. Second warning awaits a gcc LTO fix beyond our control, so closing this.

closed 2018-01-09 by dhalbert