adafruit/circuitpython #314: Add driver for TSL2561 Light Sensor

caternuson, 2017-10-11:
Add driver to support the Adafruit TSL2561 Digital Luminosity/Lux/Light Sensor Breakout.
- [PID 439](https://www.adafruit.com/product/439)

tannewt, 2017-10-11:
There is an older MicroPython version you can start from here: https://github.com/adafruit/micropython-adafruit-tsl2561

I believe @KurticusMaximus was also looking to cookiecutterify a similar repo.

caternuson, 2017-10-12:
In-work version lives here:
https://github.com/caternuson/CircuitPython_TSL2561

tannewt, 2017-10-16:
Hey, here is some feedback. I can't comment easily inline. So here it is:

* How about renaming `lux` to `light` to match the doc here, which has its roots in teh unified arduino driver spec. https://circuitpython.readthedocs.io/en/1.x/docs/design_guide.html#sensor-properties-and-units
* Nitpick:
```
        if self._read_register(TSL2561_REGISTER_CONTROL) & 0x03:
            return True
        else:
return False
```
is
```
        return self._read_register(TSL2561_REGISTER_CONTROL) & 0x03
```
* Another nitpick is the `(self, )` should be `(self)`.

Overall the API looks good to me! I ❤️ properties.

caternuson, 2017-10-16:
- fine by me. will do.
- indeed. you're better at this than i am. :)
- yah, i like doing no arg functions that way for some reason. easy to not do it though.

caternuson, 2017-10-16:
`return self._read_register(TSL2561_REGISTER_CONTROL) & 0x03`
That'd be functional, but I want to return the bool. Otherwise, you get this:
```
>>> tsl.enabled
3
```
I'd rather that say True

tannewt, 2017-10-23:
Ok, I've forked this into [Adafruit](https://github.com/adafruit/Adafruit_CircuitPython_TSL2561) and set up ReadTheDocs. Mind creating a PR to get this in the bundle and fixing the docs?

It looks like the [API docs](https://circuitpython.readthedocs.io/projects/tsl2561/en/latest/) aren't building quite right because of const. https://readthedocs.org/projects/adafruit-circuitpython-tsl2561/builds/6166595/ Try adding `from micropython import const` to the top and using ReadTheDocs to automock the micropython module. (Other drivers do this already.

caternuson, 2017-10-25:
PR done: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/pull/32

Where do I add that import?

caternuson, 2017-10-26:
Hmmmm. Maybe back to lux instead of light?
https://github.com/adafruit/circuitpython/commit/c966976591f6397eaa7821e00a51ad20fe8ae544

tannewt, 2017-10-26:
> Hmmmm. Maybe back to lux instead of light?
> c966976

Yes please. :-) Limor just tweaked it.

caternuson, 2017-10-26:
> Try adding from micropython import const to the top 

this goes in conf.py?

tannewt, 2017-10-26:
That goes in the driver and the autodoc_mock statement goes in conf.py.

caternuson, 2017-10-27:
Looks like the docs are at least building now. But this looks a bit odd (`address=<sphinx.ext.autodoc._MockModule object>`):
https://circuitpython.readthedocs.io/projects/tsl2561/en/latest/api.html#adafruit_tsl2561.TSL2561
Here's the mod I did for readthedocs:
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561/pull/3/files
Totally new to sphinx / readthedocs, so let me know if this is wrong.

tannewt, 2017-10-31:
This looks good to me! Anything left to do?

caternuson, 2017-11-01:
Not from me. Once your PR (thanks) gets merged I take it the code is good. Then it's just a matter of this PR to get it in the bundle?
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/pull/32

caternuson, 2017-11-03:
Done:
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561/pull/4

closed 2017-11-03 by caternuson