adafruit/circuitpython #315: ASF4 update and patching dataflow proposal

dhalbert, 2017-10-11:
Here's a straw proposal for how to manage keeping ASF4 up to date, patching its bugs, and adding enhancements.

1. Create a new `adafruit/asf4` repo. In branch `master`, commit the latest downloaded/scraped snapshot of all the code we might want. Top-level dir is `asf4`, with device/family-specific directories as necessary: `asf4/samd21`, `asf4/same21`, `asf4/samd51j`, or whatever. (I don't know how ASF4 distinguishes variants that are only pin changes.)
2. Create branch `master-bugfixes`. Commit changes that are bugs found and fixed by us.
3. Create project branches, like `circuitpython` or `seesaw`. They will derive from `master-bugfixes`. Commit changes in project branchesthat are enhancements, such as the additions @tannewt made for internal flash filesystems.
4. As necessary, update `master`, and then merge its changes into `master-bugfixes` and subsequently into the project branches.

The current `asf4` dirs in `adafruit/circuitpython` will become a submodule pointing to the `circuitpython` branch of `adafruit/asf4`.

This will avoid having to maintain explicit patches and keep a good history of what was changed. Essentially we are creating the missing repo that Microchip might have provided.

This basic isn't restricted to ASF4 and Atmel START. Something similar could be used to keep the nordic nRF upstream releases organized and patched. There we'll probably just get zip/tar files to update `master`, like the older ASF3 releases.

tannewt, 2017-10-11:
Sounds good to me!

dhalbert, 2017-10-11:
I could go ahead and set this up. I'll set up the initial `master`  with your `asf_update.py`, then create a `master-bugfixes` branch, and then a `circuitpython` branch. I'll apply the patches to generate changes in `circuitpython`. Are there patches that are bug fixes now, or are they all enhancements?

Is the stuff in `asf4_conf` specific to CircuitPython or should it go in the new repo?

Then I'll submit a pull request to change the `asf4` tree in `adafruit/circuitpython` to replace that tree with a submodule reference.

tannewt, 2017-10-11:
asf4_conf is specific to our set up. I'd leave it out.

Your plan sounds good!

dhalbert, 2017-10-12:
Done in #318

closed 2017-10-12 by dhalbert