adafruit/circuitpython #317: Add support for M0 Adalogger SD Card

Andon-A, 2017-10-11:
On the main page, it mentions that the Adalogger's SD card is currently not supported.

Via the Pins chart on [this page](https://github.com/adafruit/circuitpython/tree/master/atmel-samd) it appears that the SD_CS pin is linked to a pin not used by the rest of the Feather M0 versions - PA08, which is used as D4 on the Metro M0 Express.

Andon-A, 2017-10-11:
In the [pins.c file of the Feather M0 Adalogger](https://github.com/adafruit/circuitpython/blob/master/atmel-samd/boards/feather_m0_adalogger/pins.c), it looks like SD_CS is listed as PA08.

I don't have an M0 Adalogger Feather to test with, though.

tannewt, 2017-10-12:
So, the task is to correct the [README.md](https://github.com/adafruit/circuitpython/blob/master/README.md) file so it says the SD card is supported using the [library](https://github.com/adafruit/Adafruit_CircuitPython_SD).

ladyada, 2017-10-17:
woops this is a dupe of 
https://github.com/adafruit/circuitpython/issues/323#issuecomment-337007819
andon-A we've marked this as a beginner project if you'd like to try! :)

Andon-A, 2017-10-17:
@ladyada Looks quite simple. I'll probably poke at this and #323 after re-learning how to work GitHub correctly. I've got a M0 Adalogger on the way because I don't believe in leaving things untested.

ladyada, 2017-10-17:
cool!

Andon-A, 2017-11-27:
This is addressed in PR #461 - Marking as closed.

closed 2017-11-27 by Andon-A