adafruit/circuitpython #321: Audit and fix native API for methods that accept and ignore extra args.

tannewt, 2017-10-13:
Python methods can take two type of arguments:

- Positional args that depend on their position in the list and must be given.
- Keyword args are optional and either depend on position or can be identified by their name.

For example, [`audioio.AudioOut`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/audioio/AudioOut.html#audioio.AudioOut) takes two positional arguments, `pin` and `sample_source` and no keyword arguments.

In contrast, [`AudioOut.play`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/audioio/AudioOut.html#audioio.AudioOut.play) takes one keyword argument `loop` that defaults to `False` and can be positionally specified (`a.play(True)`) or specified by name (`a.play(loop=True)`).

There is a common bug in CircuitPython related to arguments where some methods defined in [`shared-bindings`](https://github.com/adafruit/circuitpython/tree/master/shared-bindings) accept extra random arguments (both positional and keyword) without a fuss (raising `TypeError: extra keyword arguments given` or `TypeError: foo() takes at most # arguments`). This can lead to great confusion! Normally Python will check argument validity for us but when we implement Python using C it trusts us to check.

There are a couple reasons this happens that need to be checked and fixed:

Constructors (`*make_new` in C and equivalent to `__init__` in Python), like `audioio.AudioOut`, should use [`mp_arg_check_num`](https://github.com/adafruit/circuitpython/blob/master/py/argcheck.c#L33) to valid the number of arguments they receive. The last argument to `mp_arg_check_num` indicates whether keyword arguments should be accepted. This is often wrong and should be fixed. The acceptable number of positional arguments (args 3 and 4) can also be wrong!

Method arguments can be checked when methods are declared using a macro that can restrict the number of arguments such as:
- [`MP_DEFINE_CONST_FUN_OBJ_2`](https://github.com/adafruit/circuitpython/blob/master/py/obj.h#L289) which restricts to 2 positional arguments and 0 keyword arguments
- [`MP_DEFINE_CONST_FUN_OBJ_KW`](https://github.com/adafruit/circuitpython/blob/master/py/obj.h#L301) which enforces a minimum number of arguments (but no max!) in the second argument and allowing for keyword arguments.
- [`MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN `](https://github.com/adafruit/circuitpython/blob/master/py/obj.h#L298) which enforces a range of positional arguments and no keyword args.
- [`MP_DEFINE_CONST_FUN_OBJ_VAR`](https://github.com/adafruit/circuitpython/blob/master/py/obj.h#L295) which only enforces a minimum number of positional arguments and no keyword args.

These macros can be misused and unintentionally allow more arguments than those used. Picking the best macro for each function is the first step to getting method argument checking correct. If the macro is wrong, the method may not catch extra positional arguments.

So, the task is to look through `shared-bindings` and validate that the argument checking matches the arguments in the docs. If it doesn't, then fix it!

C47D, 2019-07-22:
@tannewt, Does this still need fixing?

tannewt, 2019-07-22:
I don't think it is an issue if `mp_arg_parse_all` is used. I do believe there are issues within shared-bindings still. It should be easy to check, try passing a kwarg that has a weird name should error out.

I glanced at a few APIs and didn't see any issues. Fuzzing might be the easiest way to find these issues.

C47D, 2019-07-29:
Hi @tannewt,

I will not be able to work on this on short term, but I will try to fully understand the problem and solution. Fuzzing sounds like fun, I have not done it before, I gues I can build CP for Unix and fuzz it there.

jepler, 2019-07-29:
@c47d I'm not sure how I would apply fuzzing to this particular class of problem, but there's a blog post I wrote on how to apply fuzzing to find crashing bugs in circuitpython: https://www.unpythonic.net/2018/03/26/Fuzz-testing-CircuitPython.html

C47D, 2019-07-29:
@jepler Thanks, i will take a look at it on lunch time :)

jposada202020, 2021-03-13:
@tannewt 
If this is still open and valid after two yeas I am willing to work on this for the long run.  I made a list, and there are approx 680 occurrences of the macros.
list is here https://github.com/jposada202020/mp_func_macro/blob/master/test.txt

Develop the code to verify this automatically will be good, but at first sight I am not seeing as a trivial thing, maybe that will change along the way. The easiest way is to create check rules parsing the MP_  and then verifying the docs. 
I see in some of the C code that we include the python code if this is verified then the work will be easier.
Let me know. Thanks.

jposada202020, 2021-03-15:
I did a preliminary review of some of the docs vs code:
I take a look at these libraries. order dictated by commit and oswalk function in python
* touchio
* rtc
* alarm
* display
* pulseio
* bitmaptools

I found that ``MP_DEFINE_CONST_FUN_OBJ_KW`` is used differently in depending on the library
### bitmaptools: https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/bitmaptools/__init__.c#L243
### bitmap: https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/displayio/Bitmap.c#L274

Some clases using ``*make_new``does not have these ``mp_arg_check_num`` 
### pinalarm https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/alarm/pin/PinAlarm.c#L73
### timealarm https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/alarm/time/TimeAlarm.c#L71
### pulsein https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/pulseio/PulseIn.c#L86
### display https://github.com/adafruit/circuitpython/blob/3cbff45f9aac5ea2855bbc888083d356a91c80fe/shared-bindings/displayio/Display.c#L155

#### Just to be clear, just to verify that is not just me looking at things that are actually ok.

tannewt, 2021-03-16:
I believe this still may be an issue but I'm not sure with what scope. Its been a long time since I was thinking about it. I think a simple way to test could be to try and pass extra positional or keyword arguments to the APIs to see if they fail.

tannewt, 2021-03-16:
It hasn't come up recently, so the simplest thing may be to close it.

jposada202020, 2021-03-16:
I propose this:
1. I will review the shared bindings in order to verify, the class definitions, a python "lexer" house-made could help here to compare with the documentation and the enum, and mp_check..
2. Review the docs at the same time to see if we need add usage examples or correct return types.
whether the work is completed or not we close this in the summer 2021.  I will do commits with the progress.

tannewt, 2021-03-16:
@jposada202020 Ok, take a bit of time to start looking into it. If you don't find any issues in 3 modules then I'd suggest assuming the rest are ok. I don't want you to spend too much time on this.

jposada202020, 2021-03-16:
@tannewt sounds like a plan. I will report my findings. And put the results here, so far I have not seen anything wrong, besides a "," missing here and there or an space, but I consider this more styling. Thanks.

jposada202020, 2021-03-21:
@tannewt did not find any significant issue, there are for sure different methods to verify the arguments. As discussed I verify three module in shared bindings.  For me, we can keep as a reference and move-on.

## BUSIO

### I2C

|                     |  Enum/Allowed Args |FUN_OBJ_X   | VAR_BETWEEN| FUN_OBJ_KW  |
|---------------------|:------------------:|:----------:|:----------:|:-----------:|
|Make New             |          X         |            |            |             |
|deinit               |                    |     X      |            |             |
|exit                 |                    |            |       X     |             |
|scan                 |                    |      X     |            |             |
|try_lock             |                    |      X     |            |             |
|unlock               |                    |      X     |            |             |
|readfrom_into        |          X         |            |            |     X       |
|writeto              |          X         |            |            |     X       |
|writeto_then_readfrom|          X         |            |            |     X       |

### Onewire
|                     |  Enum/Allowed Args |FUN_OBJ_X   | VAR_BETWEEN|
|---------------------|:------------------:|:----------:|:----------:|
|Make New             |          X         |            |            |
|deinit               |                    |     X      |            |
|exit                 |                    |            |      X      |
|reset                |                    |      X     |            |
|write_bit            |                   |      X      |            |

### SPI
|                     |  Enum/Allowed Args |FUN_OBJ_X   | VAR_BETWEEN| FUN_OBJ_KW  |
|---------------------|:------------------:|:----------:|:----------:|:-----------:|
|Make New             |          X         |            |            |             |
|deinit               |                    |     X      |            |             |
|exit                 |                    |            |       X    |             |
|configure            |         X          |            |            |       X     |
|try_lock             |                    |      X     |            |             |
|unlock               |                    |      X     |            |             |
|write                |          X         |            |            |     X       |
|readinto             |          X         |            |            |     X       |
|write_readinto       |          X         |            |            |     X       |
|frequency            |                    |      X     |            |     X       |

### UART
|                     |  Enum/Allowed Args |FUN_OBJ_X   | VAR_BETWEEN| FUN_OBJ_KW  |
|---------------------|:------------------:|:----------:|:----------:|:-----------:|
|Make New             |          X         |            |            |             |
|deinit               |                    |     X      |            |             |
|exit                 |                    |            |       X    |             |
|get_baudrate         |                    |      X     |            |             |
|set_baudrate         |                    |      X     |            |             |
|get_in_waiting       |                    |      X     |            |             |
|get_timeout          |          X         |            |            |     X       |
|set_timeout          |          X         |            |            |     X       |
|reset_input_buffer   |          X         |            |            |     X       |
|parity               |          X**       |           |            |            |

## OS

### __init__
|                     |FUN_OBJ_X   | VAR_BETWEEN|
|---------------------|:----------:|:----------:|
|uname                |     X      |            |
|getcwd               |     X      |            |
|listdir              |            |       X    |
|mkdir                |      X     |            |
|remove               |      X     |            |
|rename               |      X     |            |
|stat                 |       X    |            |
|statvfs              |       X    |            |
|sync                 |       X    |            |
|urandom              |      X     |            |

## TOUCHIO

### Touchin
|                     |FUN_OBJ_X   | VAR_BETWEEN|mp_arg_check_num|
|---------------------|:----------:|:----------:|:--------------:|
|Make New             |            |            |                |
|deinit               |     X      |            |       X        |
|exit                 |            |       X    |                |
|get_value            |      X     |            |                |
|raw_value            |      X     |            |                |
|get_threshold        |      X     |            |                |
|set_threshold        |       X    |            |                |

### NOTES
*void functions not considered

** locals dict table

tannewt, 2021-03-22:
Ok, thanks for looking into this @jposada202020. I'll close this since it hasn't been an issue for many years.

closed 2021-03-22 by tannewt